cyberbotics/webots #6414: Crash (segfault) when boundingObject Group contains Solid with physics

PeterJCLaw, 2023-10-14:
**Describe the Bug**

Launching webots R2023b with the following world results in some warnings and a segfault. This world is a cut-down. I hit this while developing the larger world, which doesn't normally have warnings but did in the failure mode.

The warnings in the original world before the cut-down was just:
> Pose: A Pose node inside a 'boundingObject' can only contain one Shape or one Geometry node. The child node is ignored.

and then a line like:
```
/usr/local/bin/webots: line 105: 33888 Segmentation fault      (core dumped) "$webots_home/bin/webots-bin" "$@"
```

The warnings observed in the cut-down are different, so I'm not completely sure if the cause is the same.

**Steps to Reproduce**

``` console
$ webots --stdout --stderr --mode=pause $PWD/worlds/Arena.wbt
```

`Arena.wbt`:
``` python
#VRML_SIM R2022b utf8

WorldInfo {
  basicTimeStep 8
}
Viewpoint {}

Solid {
  name "outer"
  children [
    DEF BODY Group {
      children [
        Solid {
          name "inner"
          physics Physics {}
        }
      ]
    }
  ]
  boundingObject USE BODY
  physics Physics {}
}
```

**Expected behavior**

For other invalid worlds webots seems to emit warnings and refuse to run a simulation, but not crash. Matching that would be fine.

**System**
- webots: R2023b, other versions untested
 - Operating System: Ubuntu 22.04.3 LTS
 - Graphics Card: NVIDIA Corporation GP107 [GeForce GTX 1050 Ti]

ShuffleWire, 2024-02-20:
I don't know what your use case is, but I've been playing lately with Solid into Solid (especially with physic for each of them), and in the end, even if the simulation "could" work on smaller system, it doesn't scaled well and give many weird behavior (the internal solver seems to struggle to follows relationship and so on...)
I would advice (and it's what I've done, and happy with that) to stick to the Solid -> Joint -> Solid architecture, which Webots is much happier to work with.
Anyway, Webots should not crash, but in the end even it didn't I would not be confident in my simulation setup.
My two cents

ShuffleWire, 2024-04-30:
An assertion if falling 
```
#5  0x00007fb24a053eb2 in __GI___assert_fail
    (assertion=0x559f33c90e3c "mIsKinematic || mSolidMerger", file=0x559f33c906fc "nodes/WbSolid.cpp", line=1173, function=0x559f33c90e18 "dxBody* WbSolid::bodyMerger() const")
    at ./assert/assert.c:101
````

I'm adding a trace of the crash

```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007fb24a0a9e8f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007fb24a05afb2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007fb24a045472 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007fb24a045395 in __assert_fail_base
    (fmt=0x7fb24a1b9a90 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x559f33c90e3c "mIsKinematic || mSolidMerger", file=file@entry=0x559f33c906fc "nodes/WbSolid.cpp", line=line@entry=1173, function=function@entry=0x559f33c90e18 "dxBody* WbSolid::bodyMerger() const") at ./assert/assert.c:92
#5  0x00007fb24a053eb2 in __GI___assert_fail
    (assertion=0x559f33c90e3c "mIsKinematic || mSolidMerger", file=0x559f33c906fc "nodes/WbSolid.cpp", line=1173, function=0x559f33c90e18 "dxBody* WbSolid::bodyMerger() const")
    at ./assert/assert.c:101
#6  0x0000559f3397f100 in WbSolid::bodyMerger() const (this=0x559f3bb38860) at nodes/WbSolid.cpp:1173
#7  0x0000559f33986778 in WbSolid::displayWarning() (this=0x559f3bb38860) at nodes/WbSolid.cpp:2716
#8  0x0000559f3397aaec in WbSolid::postFinalize() (this=0x559f3bb38860) at nodes/WbSolid.cpp:487
#9  0x0000559f33798cea in WbGroup::recomputeBoundingSphere() (this=0x559f3c4a9800) at nodes/WbGroup.cpp:129
#10 0x0000559f33798700 in WbGroup::postFinalize() (this=0x559f3c4a9800) at nodes/WbGroup.cpp:98
#11 0x0000559f3381ef9d in WbMatter::postFinalize() (this=0x559f3e456c60) at nodes/WbMatter.cpp:132
#12 0x0000559f3397a4a2 in WbSolid::postFinalize() (this=0x559f3e456c60) at nodes/WbSolid.cpp:456
#13 0x0000559f33798cea in WbGroup::recomputeBoundingSphere() (this=0x559f35ae9c30) at nodes/WbGroup.cpp:129
#14 0x0000559f33798700 in WbGroup::postFinalize() (this=0x559f35ae9c30) at nodes/WbGroup.cpp:98
#15 0x0000559f336c9b39 in WbBaseNode::finalize() (this=0x559f35ae9c30) at nodes/WbBaseNode.cpp:118
#16 0x0000559f3395eeec in WbSimulationWorld::WbSimulationWorld(WbTokenizer*) (this=0x559f3e4bb7c0, tokenizer=0x7ffe73628a90) at engine/WbSimulationWorld.cpp:105
#17 0x0000559f3372ac30 in WbControlledWorld::WbControlledWorld(WbTokenizer*) (this=0x559f3e4bb7c0, tokenizer=0x7ffe73628a90) at control/WbControlledWorld.cpp:44
#18 0x0000559f3369c006 in WbApplication::loadWorld(QString, bool, bool) (this=0x559f35ba1c00, worldName=..., reloading=false, isLoadingAfterDownload=true) at app/WbApplication.cpp:288
#19 0x0000559f3369e6d5 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1, 2>, QtPrivate::List<QString const&, bool, bool>, void, void (WbApplication::*)(QString, bool, bool)>::call(void (WbApplication::*)(QString, bool, bool), WbApplication*, void**)
    (f=(void (WbApplication::*)(class WbApplication * const, class QString, bool, bool)) 0x559f3369b4de <WbApplication::loadWorld(QString, bool, bool)>, o=0x559f35ba1c00, arg=0x7ffe736291b0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:135
#20 0x0000559f3369e1ad in QtPrivate::FunctionPointer<void (WbApplication::*)(QString, bool, bool)>::call<QtPrivate::List<QString const&, bool, bool>, void>(void (WbApplication::*)(QString, bool, bool), WbApplication*, void**)
    (f=(void (WbApplication::*)(class WbApplication * const, class QString, bool, bool)) 0x559f3369b4de <WbApplication::loadWorld(QString, bool, bool)>, o=0x559f35ba1c00, arg=0x7ffe736291b0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:172
#21 0x0000559f3369dd7f in QtPrivate::QSlotObject<void (WbApplication::*)(QString, bool, bool), QtPrivate::List<QString const&, bool, bool>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x559f3e3aa480, r=0x559f35ba1c00, a=0x7ffe736291b0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:383
#22 0x00007fb24cf97772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#23 0x0000559f33b804e5 in WbProtoManager::worldLoadCompleted(QString const&, bool, bool) (this=0x559f360e8250, _t1=..., _t2=false, _t3=true) at build/debug/WbProtoManager.moc.cpp:203
#24 0x0000559f338d41be in WbProtoManager::loadWorld() (this=0x559f360e8250) at vrml/WbProtoManager.cpp:418
#25 0x0000559f338df79f in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbProtoManager::*)()>::call(void (WbProtoManager::*)(), WbProtoManager*, void**)
    (f=(void (WbProtoManager::*)(class WbProtoManager * const)) 0x559f338d3d84 <WbProtoManager::loadWorld()>, o=0x559f360e8250, arg=0x7ffe736293d0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:135
#26 0x0000559f338dd6d5 in QtPrivate::FunctionPointer<void (WbProtoManager::*)()>::call<QtPrivate::List<>, void>(void (WbProtoManager::*)(), WbProtoManager*, void**)
    (f=(void (WbProtoManager::*)(class WbProtoManager * const)) 0x559f338d3d84 <WbProtoManager::loadWorld()>, o=0x559f360e8250, arg=0x7ffe736293d0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:172
--Type <RET> for more, q to quit, c to continue without paging--
[... cut ...]
#66 0x00007fb24d1df90a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#67 0x00007fb24cf44c9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#68 0x00007fb24cf4107e in QCoreApplication::exec() () at /home/yann/webots/lib/webots/libQt6Core.so.6
#69 0x0000559f337a002c in WbGuiApplication::exec() (this=0x7ffe7362b520) at gui/WbGuiApplication.cpp:288
#70 0x0000559f33a84a0e in main(int, char**) (argc=1, argv=0x7ffe7362b948) at gui/main.cpp:214

```

PeterJCLaw, 2026-04-05:
Unfortunately I can't remember any more detail than I posted to the original ticket. I assume that I worked around this somehow, though I haven't worked on the simulation in several years now.

A quality-of-life fix here would be fine IMO if this isn't expected to be supported. (Even if just a warning hint before crashing)

CoolSpy3, 2026-04-06:
> do you think there is low-hanging fruit PR that we could make here to improve the quality of life

Hmmm... Yeah, I think there might be something here. I just took a scan at this, and it looks like the actual crash is being caused while trying to issue a warning. Just placing `if(isDynamic() && !mSolidMerger) return;` at the beginning of `WbSolid::displayWarning()` fixed this for me. (I.e. "If running this method would trigger the assert......... , don't.") That indicates to me that either `displayWarning()` or `bodyMerger()` is making an incorrect assumption *somewhere*. Based on my quick scan of the code, it's possible that what should happen is just setting `mIsKinematic = true`. (But I'm not sure.)

The real "work" that has to be done here is the due-diligence to figure out where exactly the oversight is. Do we have to set `mIsKinematic` somewhere, does `bodyMerger()` or `displayWarning()` need to perform an additional check, or is it something else? In either case, what then is the appropriate condition to trigger the fix?

It'll prob be a bit before I have time to dive into this (~1 mo unless I work between other stuff [which is not uncommon]). If you're interested, feel free to take a crack at this before then.

Also @ShuffleWire, thanks for triaging some of the issues in the backlog! It's long overdue....

CoolSpy3, 2026-05-02:
**UPDATE:** (Because I have to go to sleep and I don't want to forget where I am.)
The crash is caused by the solid trying to associate itself with its parent's solid merger, which isn't initialized until after its own solid merger. It looks like this is fixed by swapping the if-statement order in `WbSolid::createOdeObjects()`. (I don't think there are any side effects from this, but I haven't run a full test-suite yet.)

However, imo, there shouldn't even be a way to achieve this under normal circumstances. Webots goes to pretty great lengths to constrain allowed bounding object values. It looks like this is normally handled by `WbNode::validate()`, but this is only called when the scene tree is loaded, which is before the USE nodes are realized.

I've been playing around with inserting a `validate` call into `WbDictionary::makeDefNodeAndUpdateDictionary`, but I'm still not getting the warning messages I expect in the debug log. So, still some work to do, but I think I have a solid handle on this now.