cyberbotics/webots #6424: [Feature Request] Add wayland support

Etaash-mathamsetty, 2023-10-29:
Basically the title, all we need to do is revert https://github.com/cyberbotics/webots/pull/940, and https://github.com/cyberbotics/webots/pull/4641. Then Webots should be able to run under wayland

omichel, 2023-10-30:
Did you test if reverting these two PRs does the job or is that just a guess? If you are sure that reverting these two PRs is the way to go, please open a new PR doing this and we will review it.

dokee39, 2024-08-16:
In the latest pre-release (Webots Nightly Build (15-8-2024)), there seems to be no corresponding file added. Is it ignored?

Etaash-mathamsetty, 2024-08-16:
> In the latest pre-release (Webots Nightly Build (15-8-2024)), there seems to be no corresponding file added. Is it ignored?

yeah it's not there, odd

Etaash-mathamsetty, 2024-08-16:
probably needs some ubuntu dependency for qt to build it or smth ?

Etaash-mathamsetty, 2024-08-16:
ok seems that the webots-qt dependency they download hasn't been updated to include wayland yet

CoolSpy3, 2025-08-17:
Wanted to provide a quick update on this for anyone who finds this issue. It looks like Webots runs under XWayland by default, but native Wayland use can be forced with `-platform wayland` (although this crashes under a snap install). In Wayland mode, I've had *mixed* results. My Ubuntu vms run fine, but my Fedora install has a graphical bug that offsets the simulation viewport.