cyberbotics/webots #6425: ElevationGrid keeps falling down...

wilhem, 2023-10-30:
**Describe the Bug**
I tried to create a simple landscape by using an ElevationGrid feature. 
Despite the fact, that the ElevationGrid is visible and reflects exactly my wishes, it falls down from the world.

**Steps to Reproduce**
Take the minimal example I already created here.
Pay attention to replace my texture (2345.jpg) with another one.

```
#VRML_SIM R2023b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"

WorldInfo {
}
Viewpoint {
  orientation 0.11335379191925489 0.39200416219618567 -0.9129532598541904 0.6134529274482173
  position -8.806747535228917 7.3072727946927465 2.6034575669569326
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Solid {
  translation 0 0 1.54
  children [
    DEF Grund Shape {
      appearance Appearance {
        texture ImageTexture {
          url [
            "2354.jpg"
          ]
        }
      }
      geometry ElevationGrid {
        height [
          0, 0, 0, 0, 0, 0.6, 0.72, 0, 0, 0.2
          0.4, 0, 0, 0, 0, 0
        ]
        xDimension 4
        yDimension 4
      }
    }
  ]
  boundingObject USE Grund
  physics Physics {
  }
}
```

**Expected behavior**
Expected is that the ElevationGrid does not fall through the ground and/or without the ground.

**Screenshots**
No screenshot needed

**System**
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.3 LTS"
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

omichel, 2023-10-31:
This bug was likely fixed in #6412. Can you test if it is still there on a [nightly build of Webots R2023b-rev1](https://github.com/cyberbotics/webots/releases)?

wilhem, 2023-11-03:
I found out that the problem was caused by the assignation of physic to the elevationgrid.
Removing it fixed the problem

closed 2023-11-03 by omichel