cyberbotics/webots #6438: scripts/install/bashrc.linux be appended to .bashrc last line if no blank line

sangshuduo, 2023-11-14:
**Describe the Bug**
When I run "cat scripts/install/bashrc.linux >> ~/.bashrc" and my .bashrc's last line is not blank, the bashrc.linux first line will be appended to the end of the last line of .bashrc and make it fail to execute.

Suggestion: 
add a blank line to the bashrc.linux to avoid the issue happening.

omichel, 2023-11-14:
Thank you for reporting this problem.
Note that since the first line of the `scripts/install/bashrc.linux` is a comment (starting with `#`), it should not prevent your last line of original bashrc to be executed nor it should prevent the first command of `scripts/install/bashrc.linux` to be executed. So, I am not sure why you experienced a problem with this.

Nevertheless, I agree that it would be cleaner to append a blank line in these files to ensure a nice result in the `~/.bashrc`. So I fixed it in #6441.

closed 2023-11-14 by omichel

sangshuduo, 2023-11-14:
Good to know it is fixed with the prompt solution.

The early issue is the error "No such file or directory" when the ### be appended to a file to be sourced.

`...
[ -f ~/.inshellisense/key-bindings.bash ] && source ~/.inshellisense/key-bindings.bash#####################################################################################################
# Uncomment / edit the following environment variables when the corresponding software is installed #
#####################################################################################################
`

PR #6441 (2023-11-14): Prepend blank line in bashrc script
Fixes #6438