cyberbotics/webots #6454: Cadshape geometry contained in Solid will not be detected by radar.

ltcstar, 2023-12-05:
**Describe the Bug**
`CadShape` geometry contained in `Solid` node will not be detected by `radar`.
Only the `Solid` **origin position** can be detected by `radar`.

**Steps to Reproduce**
1. add a `solid` node, set `radarCrossSection` to 1, add `CadShape` to children and select some model to import.
2. add a `robot` with `radar` and write suitable `controller` codes to enable radar and prints radar information to console.
3. add a `supervisor robot` with `controller` codes to make the solid node move radially towards and away from radar.
4. radar detects nothing as long as the origin point of `Solid` is **out of range**, while the geometry of `CadShape` is **in range**.

**Expected behavior**
I've tried to use **Solid - Shape - Sphere**. The result shows that the `Sphere` geometry can be detected by radar. The radar can detect something after the origin of Solid is out of range when some part of Sphere is still in range.

It's expected for radar to detect CadShape geometry.

**Screenshots**
CadShape cannot be detected by radar.
[image]

Sphere can be detected by radar.
[image]

**System**
 - Operating System: Windows 11 Pro, 22H2, 22621.2715
 - Graphics Card: NVIDIA GeForce RTX 4060 8 GB

ShuffleWire, 2024-04-29:
I would expect that Webots would have to check for all individual vertexes / faces in your shape to determine accurately if it's visible or not, and that would be expensive...

As for collision detection, I stumble on using a bounding object made up of multiple basic built-in shape, and that work way better, in term of computational stability and cost...

existedinnettw, 2025-06-17:
I encounter similar problem with distance sensor, and I overcome it with `Mesh`.

```proto
                  children [
                    # DEF link2_visual CadShape {
                    #  url "../meshes/link2.dae"
                    # }
                    DEF LINK2_SHAPE Shape {
                      appearance PBRAppearance {
                        roughness 1.000000
                        metalness 0
                      }
                      geometry Mesh {
                        url "../meshes/link2.dae"
                      }
                    }
                  ]
```