cyberbotics/webots #6465: Issue Connecting Webots with Simulink MATLAB - Unable to Establish Communication

masoudamirkhani, 2023-12-31:
I am facing difficulties connecting Webots with Simulink MATLAB, and I am unable to establish communication between the two platforms. I have followed the documentation and relevant guidelines, but the connection does not seem to be working. Are there any specific steps, configurations, or troubleshooting tips that I might be missing? Any insights or assistance would be greatly appreciated.
please Add a full description of how to connect it to Simulink

Traveller30, 2024-02-26:
what steps did you implement?

harunkurtdev, 2025-10-20:
@Traveller30 and @masoudamirkhani 

if you sitll need, you can look

https://www.mathworks.com/matlabcentral/fileexchange/181015-webots-simulink-bridge

or

https://github.com/harunkurtdev/webots-simulink

omichel, 2026-02-24:
Yes, let's close it. @Traveller30 and @masoudamirkhani, feel free to re-open if you believe the problem is not sorted out.