cyberbotics/webots #6467: Initial location robot changed to 0,0 after reset

knmcguire, 2024-01-03:
**Describe the Bug**
I noticed with the crazyflie_apartment example, that if I reset the simulation, the bot goes from it's initial location (-5, -2) in the .wbt, to (0,0). This way the bot is stuck in the wall.

**Steps to Reproduce**
1. Start up the example world bitcraze/crazyflie_example
2. Play the simulator for a few seconds (crazyflie takes off but that is not the problem)
3. Reset the simulator with [image]
4. The crazyflies moves away from view and to (0,0), and now stuck in the wall
5. Reload the world with [image] to get it back in position

**Expected behavior**
I expect the bot to return to its initial location of (-5, -2)

**System**
 - Operating System: Ubuntu 22.04
 - Graphics Card: Nvidia Geforce 3060 Mobile

ShuffleWire, 2024-02-20:
In the Crazyfly Robot PROTO, on line 32, there is
`    translation 0 0 0.015`
Which afaik is causing the issue, and should be removed.
I'm correct ?

knmcguire, 2024-02-23:
Yes absolutely you are correct. I'll remote that immediately from the proto now with an PR. Thanks!

commit af596b46 (2024-02-26): remove position from crazyflie proto (#6511)

This should fix #6467