cyberbotics/webots #6469: Missing libcar.dylib on MacOS

SarvagyaVaish, 2024-01-04:
I am trying to run an extern controller based on instructions [here](https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-os=macos#single-simulation-and-single-local-extern-robot-controller). I'm on an Apple Silicon based Mac computer. When doing so, I ran into the following issue:

```
Traceback (most recent call last):
  File "/path/to/python/controller.py", line 21, in <module>
    driver = Driver()
  File "/System/Volumes/Data/Applications/Webots.app/Contents/lib/controller/python/vehicle/driver.py", line 55, in __init__
    Driver.loadApi()
  File "/System/Volumes/Data/Applications/Webots.app/Contents/lib/controller/python/vehicle/driver.py", line 49, in loadApi
    ctypes.cdll.LoadLibrary(os.path.join(os.environ['WEBOTS_HOME'], path, car))
  File "/Users/...username.../.pyenv/versions/3.10.12/lib/python3.10/ctypes/__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "/Users/...username.../.pyenv/versions/3.10.12/lib/python3.10/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller/libcar.dylib, 0x0006): tried: '/System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller/libcar.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller/libcar.dylib' (no such file), '/System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller/libcar.dylib' (no such file)
Exception ignored in: <function Driver.__del__ at 0x10569ac20>
Traceback (most recent call last):
  File "/System/Volumes/Data/Applications/Webots.app/Contents/lib/controller/python/vehicle/driver.py", line 65, in __del__
AttributeError: 'NoneType' object has no attribute 'wbu_driver_cleanup'
```

Seems like `libcar.dylib` is missing at `/System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller`. 

However, I did find this and other libs at `/System/Volumes/Data/Applications/Webots.app/Contents/lib/controller`. Once I copied over the libs as follows, things seemed to work just fine 🎉 

```
cp -r /System/Volumes/Data/Applications/Webots.app/Contents/lib/controller/* /System/Volumes/Data/Applications/Webots.app/Contents/MacOS/lib/controller
```

In other words, there seems to be an extra `.../MacOS/...` in the path. Thoughts?

**System**
 - Operating System: macOS Ventura, Apple M2 Pro
 - Graphics Card: Built in

omichel, 2024-01-05:
The launcher adds this `/System/Volumes/Data/Applications/Webots.app/Contents/lib/controller` folder in the `DYLD_LIBRAY_PATH`, see https://github.com/cyberbotics/webots/blob/master/src/controller/launcher/webots_controller.c#L392 which works fine for the C/C++ versions. However, the Python driver uses a different path as seen in https://github.com/cyberbotics/webots/blob/master/lib/controller/python/vehicle/driver.py#L46. So, I guess the bug is there and we should remove the 'MacOS' folder from there. Can you check if that fixes the issue for you?

SarvagyaVaish, 2024-01-05:
Yup. That should work for my system configuration. Not sure how to test it. 
Also, would removing `MacOS` from the path break any other system configs?

omichel, 2024-01-12:
No, you don't need to build from sources, you can simply modify this `driver.py` file from your local Webots installation. It is located somewhere in the Webots.app bundle, I sure you will find it.

SarvagyaVaish, 2024-01-17:
We'll need to update all of these locations, right?

```
/Applications/Webots.app/Contents/lib/controller/python/vehicle/car.py:
   42              path = os.path.join('lib', 'controller', 'car.dll')
   43          elif sys.platform == 'darwin':
   44:             path = os.path.join('Contents', 'MacOS', 'lib', 'controller', 'libcar.dylib')
   45          self.api = ctypes.cdll.LoadLibrary(os.path.join(os.environ['WEBOTS_HOME'], path))
   46          self.api.wbu_car_get_front_wheel_radius.restype = ctypes.c_double

/Applications/Webots.app/Contents/lib/controller/python/vehicle/driver.py:
   44              driver = 'driver.dll'
   45          elif sys.platform == 'darwin':
   46:             path = os.path.join('Contents', 'MacOS', 'lib', 'controller')
   47              car = 'libcar.dylib'
   48              driver = 'libdriver.dylib'

/Applications/Webots.app/Contents/MacOS/lib/controller/python/vehicle/car.py:
   42              path = os.path.join('lib', 'controller', 'car.dll')
   43          elif sys.platform == 'darwin':
   44:             path = os.path.join('Contents', 'MacOS', 'lib', 'controller', 'libcar.dylib')
   45          self.api = ctypes.cdll.LoadLibrary(os.path.join(os.environ['WEBOTS_HOME'], path))
   46          self.api.wbu_car_get_front_wheel_radius.restype = ctypes.c_double

/Applications/Webots.app/Contents/MacOS/lib/controller/python/vehicle/driver.py:
   44              driver = 'driver.dll'
   45          elif sys.platform == 'darwin':
   46:             path = os.path.join('Contents', 'MacOS', 'lib', 'controller')
   47              car = 'libcar.dylib'
   48              driver = 'libdriver.dylib'
```

omichel, 2024-01-17:
We should probably get rid of this folder: `/Applications/Webots.app/Contents/MacOS/lib/controller` as it seems redundant...

closed 2024-01-22 by SarvagyaVaish