cyberbotics/webots #6474: `getFirstFinalizedProtoInstance()` returns `NULL` for DEF with Lua code in v2023b

galou, 2024-01-11:
**Describe the Bug**
It seems that something is wrong with `DEF` with Lua code. We have a node defined like this
```
DEF %{= "LED_" .. name_suffix }% LED {
...
}
```

For this node, `node->getFirstFinalizedProtoInstance();` returns `NULL`.

This happens with v2023b but was working with v2023a.

**Steps to Reproduce**
1. Use Lua code in `DEF`
2. In your Python controller, call `getPosition()`
3. See the log
```
WARNING: Pose  > DEF LASER_EMITTER LaserEmitter: Cannot get the PROTO instance for node 'LED_XMINUS_1' of type 'LED'.
WARNING: Pose  > DEF LASER_EMITTER LaserEmitter: wb_supervisor_node_get_position() can exclusively be used with Pose (or derived).
```
4. `getPosition()` returns `[nan, nan, nan]`.

**Expected behavior**
It should work as in used to do in v2023a.

**System**
 - Operating System: Linux Ubuntu 22.04
 - Graphics Card: irrelevant

I'm ready to solve the issue but need some hint where to start.

omichel, 2024-01-11:
Unfortunately, I have no idea on how to fix this. I would recommend the following to investigate:
1. Check if the problem is the same with a Javascript proto (instead of Lua).
2. Check the [change log](https://www.cyberbotics.com/doc/reference/changelog-r2023) of R2023b to identify which PR may have caused this bug.

galou, 2024-01-11:
Candidates: #5978, #5676, #5774.

galou, 2024-01-11:
A PROTO with Javascript rather than Lua suffers from the same symptoms.

galou, 2024-01-11:
Can I debug the generated PROTO files (after going through the engine)?

galou, 2024-01-12:
When trying to write a small example to reproduce the issue, I wrote this PROTO:
```vrml
#VRML_SIM R2023b utf8
#     Robot {
#       children [
#         DEF led LED {
#         }
#       ]
#       controller "my_supervisor"
#       supervisor TRUE
#     }
# 

PROTO MyRobot [
  field SFVec3f translation 0 0 0
  field SFRotation rotation 0 0 1 0
  field SFString name "robot"
]
{
  Robot {
    translation IS translation
    rotation IS rotation
    name IS name
    controller "my_supervisor"
    supervisor TRUE
    children [
        DEF %{= "led" }% LED {
        }
    ]
  }
}
```
with this supervisor:
```python
"""my_supervisor controller."""

# You may need to import some classes of the controller module. Ex:
#  from controller import Robot, Motor, DistanceSensor
from controller import Supervisor

class Test(Supervisor):

    def __init__(self):
        super().__init__()

    def run(self):
        my_led = self.getFromDef('my_led')
        led = my_led.getFromProtoDef('led')
        print(f'{led=}')
        print(f'{led.getPosition()=}')

controller = Test()
controller.run()

```
and this world
```vrml
#VRML_SIM R2023b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/26e08bbf6049f8d3ec3af3dd874e5f5d4062ab96/projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/26e08bbf6049f8d3ec3af3dd874e5f5d4062ab96/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "../protos/MyRobot.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.5773502691896257 0.5773502691896257 0.5773502691896257 2.0944
  position 0 0 10
}
TexturedBackground {
}
TexturedBackgroundLight {
}
DEF transform Pose {
  children [
    DEF my_led MyRobot {
    }
  ]
}
```
but this combination does NOT trigger the issue. I'll have to dig farther.

galou, 2024-01-12:
I'm trying to understand the code. What is a PROTO parameter node?

omichel, 2024-01-12:
I am not sure, but it could be the nodes that are passed as PROTO parameters.

galou, 2024-01-12:
Do you mean a parameter whose type is a node?

omichel, 2024-01-12:
Yes.

galou, 2024-01-12:
`git bisect` allowed to determine that the bug was introduced in b95a7360fddc9fa5024f1c51facfce35a5d3e7c7 "Refactor node and PROTO creation (#5779)"

omichel, 2024-01-12:
Nice catch! I would suggest you to investigate commit-per-commit on #5779 to understand which commit did introduce the bug. Unfortunately @stefaniapedrazzi is not working any more with us and won't be able to help you on this.