cyberbotics/webots #6478: Webots cannot startup on windows

logeexpluoqi, 2024-01-15:
Click the webots startup icon, using admin, but it also crashed in a few miniutes

ERROR: Could not listen to extern controllers and robot windows on port 1244. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1243. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1242. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1241. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1240. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1239. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1238. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1237. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1236. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1235. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
ERROR: Could not listen to extern controllers and robot windows on port 1234. Cannot set the server in listen mode: 杩欎 釜鍦板潃琚繚鎶や簡. Giving up.
webots: failed to open TCP server in the port range [1234-1244]

q-qo-o, 2024-04-17:
This might be because you have Hyper-V enabled, causing Windows to reserve these ports (1234-1244).
To check the system-reserved TCP ports, run the following command in the console in administrator mode:
```cmd
netsh int ipv4 show dynamicport tcp
```
If it includes ports 1234-1244, you can change the reserved ports to 49152-65535 by running the following command in the console in administrator mode:
```cmd
netsh int ipv4 set dynamic tcp start=49152 num=16384
```
Reference link:
https://github.com/shadowsocks/shadowsocks-windows/issues/2171#issuecomment-603119696

logeexpluoqi, 2024-04-18:
oh, thanks. I will make a test soon