cyberbotics/webots #6481: stb_image.h path inclusion for using shared libraries

Kreijstal, 2024-01-17:
the file src/webots/nodes/WbBackground.cpp uses stb_image.h instead of stb/stb_image.h this is if you package webots for distros where stb is provided :)
https://github.com/Overv/VulkanTutorial/issues/251

omichel, 2024-01-19:
I would prefer to avoid creating some `webots/src/stb/stb` folder as it doesn't look very nice.

Kreijstal, 2024-01-19:
we could also change the line in `src/webots/Makefile` of
`STB_INCLUDE  = -isystem $(WEBOTS_PATH)/src/stb`
to `STB_INCLUDE  = -isystem $(WEBOTS_PATH)/src`

but I think the nested stb is nicer, the reason is that on 2018 there was no package in most systems, but now there is so systems like windows (msys2) you can simply just use [mingw-w64-x86_64-stb](https://packages.msys2.org/package/mingw-w64-x86_64-stb), on ubuntu/debian there is libstb and libstb-dev
and it compiles as it is without subcloning just using make :) (of course there is [mingw-w64-x86_64-glm](https://packages.msys2.org/package/mingw-w64-x86_64-glm?repo=mingw64) too)

omichel, 2024-01-22:
OK, then we should probably remove this stb folder and use the one from the system. We should add it to the dependencies, so that the right version gets installed:

- https://github.com/cyberbotics/webots/blob/master/scripts/install/msys64_installer.sh
- https://github.com/cyberbotics/webots/blob/master/scripts/install/linux_compilation_dependencies.sh
- https://github.com/cyberbotics/webots/wiki/macOS-installation