cyberbotics/webots #6485: Crash when I set boundingObject in a Fluid node

nm46nm, 2024-01-19:
**Describe the Bug**
I want to simulate robot in water. I find an example that tells me to do the following and the webots crashed after step 4.

**Steps to Reproduce**
1. Creat a new project
2. Add a node "fluid"
3. Set the boundingObject of the fluid as Shape
4. Set the geometry of the boundingObject as Sphere

**Expected behavior**
I have tried other node like box, but it still crashed. 

**Hardware&System**
I have tried on four computers. 
The most strong one is with intel Xeon(R) platinum 8163, 2.5Ghz*96,  Nvidia Geforce RTX4090,  Linux Ubuntu 18.04.
Another is with intel i7-10875H, Nvidia Geforce RTX2060, Windows 10.

ShuffleWire, 2024-03-19:
Can confirm on Debian 12, with Webots 2023b
But work when I use the PROTO Torus as a boundingObject, for instance
Traceback : 

```
Thread 1 "webots-bin" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) where
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007f795e2a9e8f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007f795e25afb2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f795e245472 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007f795e4b9746 in dDebug(int, char const*, ...) (num=2, msg=0x7f795e5f9740 "geom must be placeable in %s()") at ode/src/error.cpp:96
#5  0x00007f795e47dd69 in dGeomSetPosition_ST(dxGeom*, double, double, double) (g=0x55be07e09920, x=0, y=0, z=0) at ode/src/collision_kernel.cpp:569
#6  0x00007f795e5083c2 in dGeomSetPosition_MT(dxGeom*, double, double, double, void (*)(dxGeom*, double, double, double))
    (g=0x55be07e09920, x=0, y=0, z=0, _geomSetPositionFunc=0x7f795e47dcf3 <dGeomSetPosition_ST(dxGeom*, double, double, double)>) at ode/src/ode_MT/ode_MT.cpp:501
#7  0x00007f795e47defd in dGeomSetPosition(dxGeom*, dReal, dReal, dReal) (g=0x55be07e09920, x=0, y=0, z=0) at ode/src/collision_kernel.cpp:595
#8  0x000055be06b8d483 in WbGeometry::setOdePosition(WbVector3 const&) (this=0x7f79540071e0, translation=...) at nodes/WbGeometry.cpp:522
#9  0x000055be06c18037 in WbMatter::updateOdePlaceableGeomPosition(dxGeom*) (this=0x55be0e604470, g=0x55be07e09920) at nodes/WbMatter.cpp:549
#10 0x000055be06c18329 in WbMatter::updateOdeGeomPosition(dxGeom*) (this=0x55be0e604470, g=0x55be07e09920) at nodes/WbMatter.cpp:589
#11 0x000055be06c172e1 in WbMatter::createOdeGeomFromInsertedShapeItem() (this=0x55be0e604470) at nodes/WbMatter.cpp:393
#12 0x000055be06b8775f in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbMatter::*)()>::call(void (WbMatter::*)(), WbMatter*, void**)
    (f=(void (WbMatter::*)(class WbMatter * const)) 0x55be06c16f20 <WbMatter::createOdeGeomFromInsertedShapeItem()>, o=0x55be0e604470, arg=0x7fff26fa4af0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:135
#13 0x000055be06b87655 in QtPrivate::FunctionPointer<void (WbMatter::*)()>::call<QtPrivate::List<>, void>(void (WbMatter::*)(), WbMatter*, void**)
    (f=(void (WbMatter::*)(class WbMatter * const)) 0x55be06c16f20 <WbMatter::createOdeGeomFromInsertedShapeItem()>, o=0x55be0e604470, arg=0x7fff26fa4af0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:172
#14 0x000055be06b8757b in QtPrivate::QSlotObject<void (WbMatter::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)
    (which=1, this_=0x7f7954009290, r=0x55be0e604470, a=0x7fff26fa4af0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:383
#15 0x00007f7961197772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#16 0x000055be06f76cb9 in WbShape::geometryInShapeInserted() (this=0x55be0f043610) at build/debug/WbShape.moc.cpp:260
#17 0x000055be06d3ec36 in WbShape::createOdeGeom() (this=0x55be0f043610) at nodes/WbShape.cpp:476
#18 0x000055be06c1955f in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbShape::*)()>::call(void (WbShape::*)(), WbShape*, void**)
    (f=(void (WbShape::*)(class WbShape * const)) 0x55be06d3ebc8 <WbShape::createOdeGeom()>, o=0x55be0f043610, arg=0x7fff26fa4ce0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:135
#19 0x000055be06c193ff in QtPrivate::FunctionPointer<void (WbShape::*)()>::call<QtPrivate::List<>, void>(void (WbShape::*)(), WbShape*, void**)
    (f=(void (WbShape::*)(class WbShape * const)) 0x55be06d3ebc8 <WbShape::createOdeGeom()>, o=0x55be0f043610, arg=0x7fff26fa4ce0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:172
#20 0x000055be06c19237 in QtPrivate::QSlotObject<void (WbShape::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)
    (which=1, this_=0x55be0ef60480, r=0x55be0f043610, a=0x7fff26fa4ce0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:383
#21 0x00007f7961197772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#22 0x000055be06f82569 in WbValue::changed() (this=0x55be0f03d2a0) at build/debug/WbValue.moc.cpp:189
#23 0x000055be06d3abd1 in WbSFNode::setValue(WbNode*) (this=0x55be0f03d2a0, node=0x7f79540071e0) at vrml/WbSFNode.cpp:79
#24 0x000055be06c714c5 in WbNodeOperations::initNewNode(WbNode*, WbNode*, WbField*, int, bool, bool)
    (this=0x55be07e434e0, newNode=0x7f79540071e0, parentNode=0x55be0f043610, field=0x55be0f0f98e0, newNodeIndex=0, subscribe=false, finalize=true) at nodes/utils/WbNodeOperations.cpp:251
#25 0x000055be06d2a5c1 in WbSceneTree::addNew() (this=0x55be07edd7c0) at scene_tree/WbSceneTree.cpp:994
#26 0x000055be06d2cb53 in WbSceneTree::handleDoubleClickOrEnterPress() (this=0x55be07edd7c0) at scene_tree/WbSceneTree.cpp:1514
#27 0x000055be06d31109 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbSceneTree::*)()>::call(void (WbSceneTree::*)(), WbSceneTree*, void**)
[... cut ...]
#50 0x00007f79613df90a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#51 0x00007f7961144c9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#52 0x00007f796114107e in QCoreApplication::exec() () at /home/yann/webots/lib/webots/libQt6Core.so.6
#53 0x000055be06b992a2 in WbGuiApplication::exec() (this=0x7fff26fa6000) at gui/WbGuiApplication.cpp:288
#54 0x000055be06e733a8 in main(int, char**) (argc=1, argv=0x7fff26fa6418) at gui/main.cpp:214
```

ShuffleWire, 2024-03-19:
Ok got it : A sphere is considered "NON-PLACEABLE" by the ODE solver, it means that it position can't be changed (and indeed in Webots a Sphere don't have any location/rotation)  (http://ode.org/wiki/index.php, see Geoms)

The workaround in the first add a Pose as the BoundingObject, and then insert a Shape in it children, and finally add the Sphere.

I think Webots should clearly document this behavior and/or add a warning to catch that before ODE raise an abrupt ABORT signal.

CoolSpy3, 2024-12-04:
This appears to be fixed in the latest nightly. It looks like this issue has the same root cause as #6675, so it was also fixed in #6691.

closed 2024-12-04 by CoolSpy3