cyberbotics/webots #6495: Webots crash on save

PhilDel32, 2024-02-08:
**Describe the Bug**
Webots crashes every time I click on 'save' button.

**Steps to Reproduce**
1. Install webots with APT or from the tarball package.
2. Launch webots from a terminal.
3.  Create a new world (the same as in webots documentation, Tutorial 1, Hands-on 1 and 2)
4. Click on 'Save'
5. The .wbt file is saved but webots crashes with the following output :
corrupted double-linked list
/usr/local/bin/webots : ligne 105 :  3936 Abandon                 (core dumped)"

**Expected behavior**
The .wbt file saved without crashing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: LMDE 6 
 - Graphics Card: NVIDIA GeForce RTX 4070
**Additional context**
The LMDE 6 install is new, the same for webots (R2023b).
It also crashes if I add a "RectangleArena" nodes or I reuse an existing example (after copying it in a directory with write permissions)

PhilDel32, 2024-04-05:
Hi,
No, I simply created a dual boot with Ubuntu 22.04 just for Webots and everything works fine on this configuration.

ShuffleWire, 2024-04-29:
I don't have idea why it could fail here. I don't know if you are familiar with the process, but it would be great if you could build Webots in debug mode, and see what gdb say about that ! 
If you are willing to do that but don't know how, I could guide you :) Otherwise, that not a big deal if you don't want to investigate, as Webots is first developed for Ubuntu in mind

PhilDel32, 2024-04-29:
Hi,
I created a new machine with LMDE 6, installed all the tools required, downloaded the sources, compiled it and guess what : No crash on saving with this executable :-)
Philippe.

ShuffleWire, 2024-04-29:
Did you checkout the branch/tag 2023b before compiling ?

PhilDel32, 2024-04-30:
it is the R2023b revision 1 version (April 29, 2024 source code)

omichel, 2024-04-30:
Do you also get no crash from the [nightly builds](https://github.com/cyberbotics/webots/releases)?

Flamisell, 2024-04-30:
I am having the same Issue with nightly builds.

PhilDel32, 2024-04-30:
I'll try to compile and test the nightly build this afternoon to see if the crash on saving disappear.

ShuffleWire, 2024-04-30:
Ok, for those how can test on this specific system, I could give help. I don't know your expertise with that, so don't be shy to ask for precision, but don't be bothered by me taking your hand :) 
The process wil allows us to see what line is causing you a crash. It could be a memory error, or an false assertion for instance. The traceback will show us where is the program at just before dying ! 

1) Checkout on whatever version you experience the crash
2) Build webots in debug mode `make debug`
3) Run webots, and check if the crash still occurs.
If no, then say me.
Else, then...
4) Restart webots, and leave it untouched
5) Get it's PID : on my machine, the easiest way is to use `pidof webots-bin`, but on your distribution it might be different
6) Attach gdb to webots : `gdb -p pid_of_the_process`
7) Type `continue`, and check if webots is still alive. (could you interact with it normally ?)
8) Try to reproduce the crash in webots
If it don't happen (webots do the operation happily) then say me.
If it happen (webots should freeze), then...
9) In gdb terminal, type `where`, and copy paste here the log it show you. Please ensure you get all the traceback, it could be very long, like 100 lines long.

PhilDel32, 2024-04-30:
Yes, I could do that but if I build webot from source on my LMDE machine, it does not crash when I save. I may have 2 options :
- Rebuild in debug mode on an Ubuntu 22.04 and copy the library and the executable on my LMDE machine and do the test like you suggest above.
- Get a version you compile for Ubuntu in debug mode and do the test on my LMDE machine. I guess this is a better option, we will avoid side effects we could have if I compile by myself.

ShuffleWire, 2024-04-30:
Whelp... webots-bin is 171MB, so not gonna fit here xD

Just for me : where did you get the buggy version you run on your LMDE ?

PhilDel32, 2024-04-30:
I got it by running the Debian installation procedure : 

sudo mkdir -p /etc/apt/keyrings
cd /etc/apt/keyrings
sudo wget -q https://cyberbotics.com/Cyberbotics.asc
echo "deb [arch=amd64 signed-by=/etc/apt/keyrings/Cyberbotics.asc] https://cyberbotics.com/debian binary-amd64/" | sudo tee /etc/apt/sources.list.d/Cyberbotics.list
sudo apt update
sudo apt install webots

ShuffleWire, 2024-04-30:
Ok then... If the version you build yourself work just fine, by all mean, that could be the solution.
Maybe running software made for Ubuntu in LMDE triggers so bad behavior, some library could be faulty, idk.

I know that Cyberbotics don't "support" others distribution than Ubuntu, so I could understand they don't want to bother with that.

Flamisell, 2024-05-02:
Nope :/ 

NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"

should I install the 2024 version?

ShuffleWire, 2024-05-06:
> Nope :/
> 
> NAME="Ubuntu" VERSION="20.04.6 LTS (Focal Fossa)"
> 
> should I install the 2024 version?

I don't think it will change the issue, but you can try

Are you able to manually build Webots from source ? If so, it would be handy if you could rebuild from source, and run the debugger on it.

Arcachofo, 2024-08-13:
Same problem here.

OS: **Linux Mint 21.1** Vera, based on Ubuntu 22.04 jammy.
Webots: **R2023b**, June 27 2023, installed from deb package downloaded from official website.

**Crash at save if "Capture thumbnail on world save or share" is activated.**
Disabling this option solves the problem.

Relevant bits of backtrace:

`       Message: Process 63725 (webots-bin) of user 1000 dumped core.
                
                Found module linux-vdso.so.1 with build-id: 14bef519b0ba51c4f5319b4e6db5b6f6db2ad708
                Found module libogg.so.0 with build-id: 842412496daf280ca5491dfb6581df5a378b72f0
                Found module libopus.so.0 with build-id: 53a29374ac557d631d863d9908ff5bec82f55ba5
                Found module libvorbisenc.so.2 with build-id: 5d30e2a713669895dcf3b298ffed2e4ce56121e8
                Found module libvorbis.so.0 with build-id: fc9c832676aac541183d1b4774746bec48b66051
                Found module libFLAC.so.8 with build-id: 280eaf0c50a65ab561e803707fca2ec714cd28a3
                Found module libapparmor.so.1 with build-id: 05135bf3304c5a7f40dfeb28b0ce953d78f1e10e
                Found module libasyncns.so.0 with build-id: f192c4e5be41e201707cb7a5a962d43de3c675d2
                Found module libsndfile.so.1 with build-id: 08e2d078caf371d274047899e3526f7e6c0cf611
                Found module libpulsecommon-15.99.so with build-id: a44864e159472d10476d2b7ef59d4038d02accc6
                Found module libpulse.so.0 with build-id: 652b88e6ada7e05945eee6af54d6af7fc2e8eb2c
                Found module libnss_sss.so.2 with build-id: e8d68d81534321ed484665f8544b27d29584311b
                Found module libnss_systemd.so.2 with build-id: aadd75edfacc2c5b9b1e467011a04c5765e4a1f8
                Found module libnss_mdns4_minimal.so.2 with build-id: a976800e191c9353f1e93a5b436c126f2d672ffb
                Found module libssl.so.1.1 with build-id: 32c6d9dfc0e4aad7ce47f7f6e63d44314fa9c1e1
                Found module libcrypto.so.1.1 with build-id: 894b27aca5ad1a9f3dd02a966aeccd15a12874a9
                Found module libqcertonlybackend.so with build-id: 832d280b0d064868b98d6234ff1fe5ee1243aaab
                Found module libqopensslbackend.so with build-id: 785af9e763c80fb67e1b638090ba7e5bcad3c098
                Found module libpciaccess.so.0 with build-id: 4c93da952f83c1d085d4bdead692db544c79b004
                Found module libtinfo.so.6 with build-id: e22ba7829a55a0dec2201a0b6dac7ba236118561
                Found module libedit.so.2 with build-id: 9fcc3490de1a1f4919b1f6e398c99c7fa25f6863
                Found module libdrm_intel.so.1 with build-id: 78e82902c2a74278a5bac31c6cb8361a78bf7586
                Found module libdrm_nouveau.so.2 with build-id: 717e881b4ee3b0bc2344b0f63c4fc10c84263f13
                Found module libdrm_amdgpu.so.1 with build-id: 55735af4f6152d6148117c98fa6db08ed1d34fe5
                Found module libelf.so.1 with build-id: 8197b90f1d133bee2975b27467a1a7268de25b63
                Found module libdrm_radeon.so.1 with build-id: 2c199d459f832a915a700e1a9234cf9e14656889
                Found module libsensors.so.5 with build-id: 0d5803327d5e32a301844fb819fd3c762f8e69f6
                Found module libLLVM-15.so.1 with build-id: 5d46c4221e9c32a1695da30bd95956dd9ce00984
                Found module radeonsi_dri.so with build-id: 52a5aed7f25cf3f3191caa1073703cdac74d3d62
                Found module libxcb-present.so.0 with build-id: a4f415af56b06f1f38d5a0339dd9828ff136b115
                Found module libxcb-dri3.so.0 with build-id: b6aa69fda711358025ae4dd29fa597866d29b8c5
                Found module libxshmfence.so.1 with build-id: f08296172ec7b676e1b371afc936e36add17366d
                Found module libXxf86vm.so.1 with build-id: a32520dd34e48cb9c4bd756f64875be8a1ef5b54
                Found module libxcb-dri2.so.0 with build-id: dc5a2c2b0c9080fff87ea241e7298a7ac2f838e3
                Found module libdrm.so.2 with build-id: 9c7cb19295d20e515902cb0710326a0b8d6394c8
                Found module libglapi.so.0 with build-id: aca6dcce398fd4458e1017c0f42bc36167e54231
                Found module libGLX_mesa.so.0 with build-id: 81f27aa4cfe213187c5a2dda6902a87fd132a76e
                Found module libxcb-glx.so.0 with build-id: 975ca99dc81fe3349892acd1ddc8d2a169810352
                Found module libqxcb-glx-integration.so with build-id: 653847ebf145aa8ff9e901477bea7cf57f304cfa
                Found module libqjpeg.so with build-id: 1f0ec50e8ff239081202d0f45d0526f2a34e81e2
                Found module libcomposeplatforminputcontextplugin.so with build-id: 0e63ed92ac5a3b35643f44f91f1046698a4dc953
                Found module libgvfscommon.so with build-id: e7fa580fb91bb21aac9fc69229e0961dc743f2b1
                Found module libgvfsdbus.so with build-id: 28abe694dac468552cc1d23d6f643e7989de85e8
                Found module libicudata.so.70 with build-id: b1c2496dd0543023c7a19c961bb7f3abc818f465
                Found module libicuuc.so.70 with build-id: bef3ff1d70aadd68aab07d858a759360c8b919ae
                Found module libxkbfile.so.1 with build-id: b585f533dd10bd8c043f887705d4e45828baf206
                Found module libxml2.so.2 with build-id: aebf8e42966c3ce475ff9d9d51a762831adcbb61
                Found module libxklavier.so.16 with build-id: a581072418d786fb4a1d160ee7143bbd977025e7
                Found module libgnomekbd.so.8 with build-id: 636c4ae86d1bff2aa2cb92b39b42e41ff714ec7c
[... cut ...]
#21 0x00007f9d0272a3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007f9d059a890a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/local/webots/lib/webots/libQt6Core.so.6
#23 0x00007f9d0570dc9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/local/webots/lib/webots/libQt6Core.so.6
#24 0x00007f9d0570a07e in QCoreApplication::exec() () at /usr/local/webots/lib/webots/libQt6Core.so.6
#25 0x00005577d6b2862d in WbGuiApplication::exec() ()
#26 0x00005577d6a1a443 in main ()
`