cyberbotics/webots #6498: Blank Space On Rendering Area

NatePolizogo, 2024-02-10:
I get black space arounf my rendering area as you can see in the picture.

**Steps to Reproduce**
I Havent found yet any way to reproduce it on any other computer, It is not version related either, I tried older versions too.
When I open webots I guess a black space around the rendering space which is very anying.

[image]

**System**
 - Operating System: Ubuntu 22.04
 - Graphics Card:  NVIDIA GeForce RTX 4070 12 GB

omichel, 2024-02-10:
This looks like this: #5024.

NatePolizogo, 2024-02-11:
I tried:
export QT_USE_PHYSICAL_DPI=1
export QT_SCALE_FACTOR=2
export QT_ENABLE_HIGHDPI_SCALING=0

Only the first one fixes partially the problem, it reduces the amount of unused space, but it's still there
I cannot click on the objects either.

[image]

NatePolizogo, 2024-02-12:
Update:

setting also QT_SCALE_FACTOR to 0.889 with some experiments to find  the right value, fixed it for me.

but I think its dirty having to export these variables each time and its a bug that needs to be fixed

Solution:
export QT_SCALE_FACTOR=0.889