cyberbotics/webots #6503: Webots shows an error message if MSYS2 is installed

Stefan248, 2024-02-16:
**Describe the Bug**
On Windows if MSYS2 with gdb is installed and added to the PATH variable webots shows the follwing error on startup:
[image]

**Steps to Reproduce**
1. Install MSYS2
2. Add C:\msys64\ucrt64\bin to the PATH variable
3. Install gdb in MSYS2 via pacman -Sy mingw-w64-ucrt-x86_64-gdb
4. Start webots -> See the error

**Additional context**
Webots tries to load libssl-3-x64.dll to check for updates. The dll is not part of the webot msys environment, therefore it tries to load the dll from PATH.

Some time ago libssl-1_1-x64.dll was part of the webot msys environment but it was removed during the port to Qt6. Maybe we should add the library again?

Another solution would be to remove MSYS2 entries from PATH under windows when webots gets started like `C:\msys64\mingw64` and `C:\msys64\ucrt64`.
In WbApplication.cpp there is already something which does this for older MSYS versions:

```
#ifdef _WIN32
  // On Windows, if Webots is started from a DOS console or from the
  // Windows desktop, we need to remove the path to msys\1.0\bin (if any)
  // to prevent the Makefile to use the mkdir.exe and rmdir.exe provided
  // by MSYS which conflict with the corresponding DOS commands (same
  // name but different syntax) and cause the Makefile to fail.
  // If Webots is started from MSYS, we shouldn't remove this path.
  QString MSYSTEM(qgetenv("MSYSTEM"));
  QString TERM(qgetenv("TERM"));
  if (MSYSTEM != "MINGW32" && TERM != "cygwin") {  // we are in the DOS or Windows Desktop case, not MSYS
    QString path(qgetenv("Path"));
    QString newPath(path);
    while (1) {
      const int i = newPath.indexOf("\\msys\\1.0\\bin", 0, Qt::CaseInsensitive);
      if (i == -1)
        break;
      int j = 0;
      for (j = i; j > 0; j--) {
        if (newPath[j] == ':')
          break;  // Volume separator
      }
      j--;  // points to volume name (e.g., "C")
      newPath = newPath.mid(0, j) + newPath.mid(i + 14);
    }
    qputenv("Path", QByteArray(newPath.toUtf8()));
  }
#endif
}
```

Kreijstal, 2024-03-27:
generally you should not add msys2 ucrt env to path, this will break not just this program.

CoolSpy3, 2024-09-21:
Resolved by #6553.

closed 2024-09-21 by CoolSpy3