cyberbotics/webots #6505: Getting "squared" point clouds with 3d lidar.

francocipollone, 2024-02-19:
**Describe the Bug**

When using lidar with a high amount of samples I get this squared format in the point cloud
[image]

**Steps to Reproduce**
1. Open 'Lidar' sample in webots
2. Modify Lidar specs using:
[image]
3. Enable point cloud visualization

**Expected behavior**
Point cloud should be computed correctly

**Screenshots**
[image]
[image]

**System**
 - Operating System: Linux Ubuntu 20.04 & 22.04(docker)
 - R2023b Webots version installed from binaries.
 - Graphics Card: RTX 3050 ti Laptop

francocipollone, 2024-02-20:
[image]

@ShuffleWire Thanks for quick reply!

ShuffleWire, 2024-02-20:
Got the same issue here...
Never used rotating lidar, so I can't really help...

TonyWelte, 2024-03-17:
I couldn't reproduce this on the master branch by manually compiling the project. Using the nightly builds versions 2023b-rev1 seem to work fine but version 2024a does appear to produce point could rotated by 180 degrees.

Looking at the diff between master and develop it appears there are some update introduced on develop
```
--- a/src/webots/nodes/WbLidar.cpp
+++ b/src/webots/nodes/WbLidar.cpp
@@ -362,7 +362,8 @@ void WbLidar::copyAllLayersToMemoryMappedFile() {
     maxWidth = ((double)w / 2.0) * (1.0 + ratio);
 
     double tmpAngle =
-      (fabs(mPreviousRotatingAngle / (2.0 * M_PI)) - floor(fabs(mPreviousRotatingAngle / (2.0 * M_PI)))) * (2.0 * M_PI);
+      (fabs((mPreviousRotatingAngle - M_PI) / (2.0 * M_PI)) - floor(fabs((mPreviousRotatingAngle - M_PI) / (2.0 * M_PI)))) *
+      (2.0 * M_PI);
     if (tmpAngle < 0)
       tmpAngle += 2.0 * M_PI;
     widthOffset = resolution * (tmpAngle / (2.0 * M_PI));
@@ -383,7 +384,7 @@ void WbLidar::copyAllLayersToMemoryMappedFile() {
         memcpy(data + (i + 1) * resolution + minWidth + widthOffset, mTemporaryImage + width() * (int)(i * skip) + minWidth,
                sizeof(float) * abs(minWidth + widthOffset));
         memcpy(data + i * resolution, mTemporaryImage + width() * (int)(i * skip) - widthOffset,
-               sizeof(float) * (maxWidth + widthOffset));
+               sizeof(float) * abs(maxWidth + widthOffset));
       }
     }
   }
```

These changes have been introduced by #6324. There must have been a conflict between the changes I made in #6417 and those of #6324, there is a missing `- M_PI` when `theta0` is computed for rotating lidars.

Note: The "tilt angle" can also produce point clouds slightly rotated #37

ShuffleWire, 2024-03-19:
That should be it ! Could you make a quick PR to fix that ?

ShuffleWire, 2024-04-30:
Closing because PR is merged

closed 2024-04-30 by ShuffleWire