cyberbotics/webots #6531: 3D window offset

Flamisell, 2024-04-04:
**Describe the Bug**
For every world (new world or default ), the default appearence of the 3D window is shifted down-left.  I tried launching webots with the  safe mod activated:

$ export WEBOTS_SAFE_MODE=true
$ webots

But the 3D window appeares off when I add a background.

**Steps to Reproduce**
$ webots

**Expected behavior**
correclty placed 3D Window.

**Screenshots**
[image]

**System**
 - Operating System:
[image]

 - Graphics Card: 
[image]

carrliitos, 2024-04-08:
This behavior also happens to me whenever I save my world with the `CTRL` + `SHIFT` + `S` command. Then I just click anywhere on the background of my world, and it fixes it for me.

Flamisell, 2024-04-09:
When I try to save my world it gives me: 

double free or corruption (out)
/usr/local/bin/webots: line 105:  6841 Aborted                 (core dumped) "$webots_home/bin/webots-bin" "$@"

And the app closes.

chengzhou-31, 2024-04-26:
I also had this issue, did you happen to find a solution to your problem?

Flamisell, 2024-04-26:
Both the window offset and the crash when saving are still not resolved.

nassir90, 2024-05-23:
I am also suffering from this issue. It seems to impact all X11 window managers, I have experimented on i3 gaps version 4.2.3, qtile 0.25.1, xfce4 version 4.18.3. It does not happen in my wayland window manager, sway.

However, I am also suffering from a huge amount of segfaults making webots effectively unusable. This may be because of arch though.

[image]

nassir90, 2024-05-24:
I am circumventing it by executing it in sway in my regular window manager. That can be done as follows

```
echo exec webots > ${config=`mktemp`} ; sway -c $config
```