cyberbotics/webots #6532: When i open empty.wbt it did not open blockly when i have it insalled.

FireCreator418, 2024-04-06:


ShuffleWire, 2024-04-29:
Please provide more detailed info on your setup (OS, webots...) The step you took and what is the result you expect to get.

And first, what is blockly ?

FireCreator418, 2024-04-29:
> Please provide more detailed info on your setup (OS, webots...) The step you took and what is the result you expect to get.
> 
> And first, what is blockly ?

Blockly is block base and it’s for making webots easy.
And i just want to ask do you use a progaming language?
If you do just tell me.

ShuffleWire, 2024-04-30:
Yes I'm using C/C++/Python

I'm not familiar with Blockly, nor how it's supposed to work, but please share a world that should work but don't.

ShuffleWire, 2024-04-30:
Ok I got the same issue : So for a repeatable path : 
1) git clone https://github.com/anadiedrichs/drone-deep-rl.git
2) Open the world empty.wbt in webots
3) We should get the same windows as https://youtu.be/bWE3aRRMNLY?t=356
[image]
But we are not.
I'm running Webots 2024a Build 15/2/2024, on Debian 12

Thing to note : AFAIK this plugins is made for Webots R2021a, so It might be expected to not work anymore.

**Solution :** 
On the tree, Find the Robot "robot" node, right click, and Select "Show robot windows"
Say me if it work for you

FireCreator418, 2024-05-09:
Thanx!

closed 2024-05-09 by ShuffleWire