cyberbotics/webots #6534: ERROR: Error downloading EXTERNPROTO 'Oak': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/trees/protos/Oak.proto', error code: 3: Host raw.githubusercontent.com not found

1642600525, 2024-04-09:
ERROR: Error downloading EXTERNPROTO 'Oak': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/trees/protos/Oak.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'CardboardBox': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/factory/containers/protos/CardboardBox.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'Moose': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/robots/clearpath/moose/protos/Moose.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'Sassafras': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/trees/protos/Sassafras.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'SandyGround': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/appearances/protos/SandyGround.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'TexturedBackgroundLight': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: Error downloading EXTERNPROTO 'TexturedBackground': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2023b/projects/objects/backgrounds/protos/TexturedBackground.proto', error code: 3: Host raw.githubusercontent.com not found
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':43:1: error: Skipped unknown 'TexturedBackground' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':45:1: error: Skipped unknown 'TexturedBackgroundLight' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':52:18: error: Skipped unknown 'SandyGround' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':72:1: error: Skipped unknown 'Sassafras' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':76:1: error: Skipped unknown 'Sassafras' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':83:18: error: Skipped unknown 'SandyGround' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':97:1: error: Skipped unknown 'Moose' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':108:1: error: Skipped unknown 'CardboardBox' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':115:1: error: Skipped unknown 'CardboardBox' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':121:1: error: Skipped unknown 'CardboardBox' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':127:1: error: Skipped unknown 'CardboardBox' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':133:1: error: Skipped unknown 'CardboardBox' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':139:1: error: Skipped unknown 'Sassafras' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':142:1: error: Skipped unknown 'Sassafras' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':146:1: error: Skipped unknown 'Sassafras' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':150:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':153:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':157:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':161:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':165:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':169:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':173:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':177:1: error: Skipped unknown 'Oak' node or PROTO.
ERROR: 'C:/Users/16426/AppData/Local/Programs/Webots/projects/robots/clearpath/moose/worlds/moose_demo.wbt':181:1: error: Skipped unknown 'Oak' node or PROTO.

1642600525, 2024-04-09:
v:2023b

omichel, 2024-04-09:
This error comes from your firewall.

closed 2024-04-09 by omichel