cyberbotics/webots #6539: webots crash

anadiedrichs, 2024-04-13:
**Describe the Bug**

Hi! I started with Crazyflie simulation in Webots a few months ago.

I want to use it to experiment with Crazyflie quadcopter and reinforcement learning. Crazyflie should take off initially, and then I would consider the following actions. 

I'm having some issues with the simulator because it crashes after the drone takes off. 

I made a video showing the behavior (https://github.com/anadiedrichs/drone-deep-rl/tree/main/1_crazyflie/controllers/v2_openaigym_stablebaselines/README.md )

**Steps to Reproduce**
 You can access to the simulation code in the repository https://github.com/anadiedrichs/drone-deep-rl/tree/main/1_crazyflie/controllers/v2_openaigym_stablebaselines

**Expected behavior**

I made a video showing the behavior (https://github.com/anadiedrichs/drone-deep-rl/tree/main/1_crazyflie/controllers/v2_openaigym_stablebaselines/README.md )

**Screenshots**

Video available in https://www.youtube.com/watch?v=HhNYX6uOy7g

**System**
 - Operating System: Linux Ubuntu 22.04.4 LTS
 - Graphics Card: Mesa Intel® UHD Graphics (ICL GT1)
 - Webots version: Webots R2023b

ShuffleWire, 2024-04-30:
I've 2 issues : 
`ModuleNotFoundError: No module named 'deepbots'`
and 
`ModuleNotFoundError: No module named 'gym'`
I could tried to test without those, but it would be easier if you could provide a stripped down version without those modules.
That would help find the issue as well.
Anyway, I do initially think that there is something bad happening in your controller, that mess up webots.
In the video you upload, you didn't show the webots console. Are there any message in those when the simulation stop ? like at https://youtu.be/HhNYX6uOy7g?t=13

ShuffleWire, 2026-04-05:
Hi, do you still have the issue ? Otherwise close it.