cyberbotics/webots #6549: stdout failed to redirect to the console

nchygy, 2024-05-22:
**Describe the Bug**
A clear and concise description of what the bug is.
  I add 3 standard output codes to the sample controller from the sample " pioneer3at.wbt ", which has been saved as another new project.
   codes are:
        printf("000000000000000stdoutIs normal");
       fprintf(stderr, "0000Error message\n");
       fprintf(stderr, "ErrorOutputIsNormal.\n");
Only "0000Error message" "ErrorOutputIsNormal." can be found in the console. That means only stderr is correctly redirect. 

**Steps to Reproduce**
1. Choose "save all necessary file to another location" when make change to the sample word "pioneer3at.wbt" and webots asking you to do that.
2. Please add the three codes before the loop as you can find in the screen shots and  run pioneer3at.wbt then check the console

**Expected behavior**
printf should work normally.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System**
 - Operating System: [Windows 11]
 - Graphics Card: [NVIDIA GeForce RTX 2050 11 GB]

**Additional context**
Add any other context about the problem here.

lukicdarkoo, 2024-05-28:
> printf("000000000000000stdoutIs normal");

@nchygy I believe you are missing `\n` at the end of the line. Webots will redirect output only once it reaches the line terminator.

lukicdarkoo, 2024-11-28:
If the issue persists feel free to reopen it.

closed 2024-11-28 by lukicdarkoo