cyberbotics/webots #6570: Fail to install webots on WSL2 Ubuntu-22.04

Roberyan, 2024-07-01:
**Describe the Bug**
Though appointing `WEBOTS_HOME`,`ros2 launch webots_ros2_universal_robot multirobot_launch.py` cmd always says "Do you want Webots R2023b to be automatically installed in C:\Program Files\Webots ([Y]es/[N]o)?" And this will fails with `[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: [Errno 2] No such file or directory: '/mnt/c/Temp/webots-R2023b_setup.exe'`  regardless the `WEBOTS_HOME`

**Cmds & results**
```
lby@Roberyan-LAPTOP-U9RB9KFV:~/ros2_ws$ ls "$WEBOTS_HOME"
webots-R2023b_setup.exe
lby@Roberyan-LAPTOP-U9RB9KFV:~/ros2_ws$ echo $WEBOTS_HOME
/mnt/d/Linux/ROS/Webots
lby@Roberyan-LAPTOP-U9RB9KFV:~/ros2_ws$ ros2 launch webots_ros2_universal_robot multirobot_launch.py
[INFO] [launch]: All log files can be found below /home/lby/.ros/log/2024-07-01-18-14-28-674988-Roberyan-LAPTOP-U9RB9KFV-80794
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: Webots directory `/mnt/d/Linux/ROS/Webots` specified in `WEBOTS_HOME` is not a valid Webots directory or is not found.
Webots R2023b was not found in your system.
- If you want to manually install Webots R2023b please download it from https://github.com/cyberbotics/webots/releases/tag/R2023b.
- If you already have Webots R2023b installed please then specify the `WEBOTS_HOME` environment variable.

Do you want Webots R2023b to be automatically installed in C:\Program Files\Webots ([Y]es/[N]o)?: n
Missing Webots version R2023b
lby@Roberyan-LAPTOP-U9RB9KFV:~/ros2_ws$ ros2 launch webots_ros2_universal_robot multirobot_launch.py
[INFO] [launch]: All log files can be found below /home/lby/.ros/log/2024-07-01-18-14-55-419132-Roberyan-LAPTOP-U9RB9KFV-81038
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: Webots directory `/mnt/d/Linux/ROS/Webots` specified in `WEBOTS_HOME` is not a valid Webots directory or is not found.
Webots R2023b was not found in your system.
- If you want to manually install Webots R2023b please download it from https://github.com/cyberbotics/webots/releases/tag/R2023b.
- If you already have Webots R2023b installed please then specify the `WEBOTS_HOME` environment variable.

Do you want Webots R2023b to be automatically installed in C:\Program Files\Webots ([Y]es/[N]o)?: y
Installing Webots R2023b... This might take some time.
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: [Errno 2] No such file or directory: '/mnt/c/Temp/webots-R2023b_setup.exe'
lby@Roberyan-LAPTOP-U9RB9KFV:~/ros2_ws$ 
```

**System**
 - Operating System: [ Windows 11, WSL2 Linux Ubuntu 22.04]
 - Graphics Card: [None]

ShuffleWire, 2024-07-02:
I'm not familiar with ROS, but I've installed Webots on wsl like you without issue. Did you install it through the apt package manager ?

Roberyan, 2024-07-02:
> I'm not familiar with ROS, but I've installed Webots on wsl like you without issue. Did you install it through the apt package manager ?

I am following the [official tutorial](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html) I tried both the apt and source file, they all fail in this step. Did you also use `WSL2` for installation?

gabryelreyes, 2024-08-07:
> '/mnt/c/Temp/webots-R2023b_setup.exe'
How does your WEBOTS_HOME look like? because it looks for me as if you were pointing to the installer but not to the installed application.

AlexanderSalge, 2024-10-29:
I think @Roberyan means he installed ROS via the [official tutorial for WSL2](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html) with a Windows install of Webots. And in order to have ROS point to the correct version of Webots (since not locally installed) you export the (mounted) windows path. But ROS still fails to find/run it

LucaFranceschi01, 2025-01-26:
Same issue here, any known fix?

szh1213, 2025-04-11:
This issue is caused by webots-ros2-driver, and if it detects that the system is a wsl system, it will try to run "webots.exe" instead of webots. You can avoid this problem by modifying the code

```bash
sudo vim /opt/ros/humble/local/lib/python3.10/dist-packages/webots_ros2_driver/utils.py
```

Find the is_wsl function, around line 85, so that modifying the code will run the webots inside WSL

```python
def is_wsl():
    return False                                                                                                        
    return 'microsoft-standard' in uname().release
```

BlueAndi, 2026-02-21:
Find the utils.py for jazzy here:
```
sudo vim /opt/ros/jazzy/lib/python3.12/site-packages/webots_ros2_driver/utils.py
```