cyberbotics/webots #6573: Node Ancestry Should be Considered for Node Type Restrictions in PROTOs

CoolSpy3, 2024-07-02:
**Describe the Bug**
At the moment, Node fields in PROTOs are required to list every type which they can accept (if they are type restricted). However, this us true even if all of the necessary types share a common base type which would be sufficient. This leads to verbose type restrictions where only a single actual type is required, and is unintuitive given the inheritance-style structure of nodes.

**Steps to Reproduce**
Attempt to load the PROTOs:
``` proto
# Base.proto
#VRML_SIM R2023b utf8

PROTO Base [] { Pose {} }
```
``` proto
# ExtendsBase.proto
#VRML_SIM R2023b utf8

EXTERNPROTO "../protos/Base.proto"

PROTO ExtendsBase [] { Base {} }
```
``` proto
# Example.proto
#VRML_SIM R2023b utf8

EXTERNPROTO "../protos/Base.proto"
EXTERNPROTO "../protos/ExtendsBase.proto"

PROTO Example [
    unconnectedField MFNode{Base{},Solid{}} f [ExtendsBase{},LightSensor{}]
]
{ Pose {} }
```

When trying to load `Example.proto`, Webots will state that the default values of `f` are not permitted even though `ExtendsBase` *is* an instance of `Base`, and `LightSensor` *is* an instance of `Solid`.

**Expected behavior**
The entire ancestry of nodes should be considered when determining if they satisfy a node restriction. This would make the above code valid.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce 1050 Ti

omichel, 2024-07-02:
That sounds like a good idea. Please go ahead with the implementation and open a new PR to propose your contribution. We will review it and likely accept it.

CoolSpy3, 2025-02-20:
Fixed in R2025a

closed 2025-02-20 by CoolSpy3