cyberbotics/webots #6597: Release 2024a

gabryelreyes, 2024-07-30:
This checklist is based on the [Release Procedure](https://github.com/cyberbotics/webots/wiki/Release-Procedure)

## Pre-Release

- [ ] Write a blog post announcing the new release.
- [ ] Decide which issues must be closed for this release.
- [ ] Close all relevant issues
- [ ] Move remaining issues to the next release
- [ ] Regenerate thumbnails with `docs/generate_thumbnails.py`
- [ ] Update changelog files for the release if necessary (e.g. merge previous and current year, etc.).
- [ ] Update all the worlds and check for unwanted warnings by running the `tests/test_worlds.py` script (may take a long time).
- [ ] If new robots/simulations are available, eventually improve the guided tour.
- [ ] Perform tests (first iterations can be done on the last nightly build of develop)
- [ ] Sort ChangeLog entries in order of importance.
- [ ] Create Package
  - [ ] Merge the develop branch into master branch.
  - [ ] Create a tag with the last commit on the Webots repository.
  - [ ] Create a tag with the last commit on the Webots-Docker repository.
  - [ ] Create a tag with the last commit on the Webots-Snap repository.
- [ ] Perform the QA test preferably on a clean computer.

## Release

- [ ] Run `scripts/new_version/new_version_file_headers.sh`
- [ ] Open a PR (from `master` in `released`) ??
- [ ] Merge the PR
- [ ] Upload the `webots/resources/web/wwi` files to https://cyberbotics.com/wwi/R2024a/
- [ ] Update Cyberbotics's web site: `webots_current_version.txt`
- [ ] Publish the snap package on the release channel on [snapcraft.io](https://snapcraft.io)
- [ ] Update the `debian` repository on [cyberbotics.com/debian](https://cyberbotics.com/debian) from a Linux machine

## Post-release

- [ ] Switch the Github tag from `draft` to `release`.
- [ ] Close the corresponding Github milestone.
- [ ] Announce the release on:
  - [ ] Twitter
  - [ ] Discord
  - [ ] LinkedIn
  - [ ] robotics-worldwide mailing list.
  - [ ] euron-dist mailing list.
  - [ ] ROS discourse.
  - [ ] Reddit.
  - [ ] HackerNews.
  - [ ] cyberbotics newsletter.
- [ ] Announce the release privately to the users waiting for it.
- [ ] Update Wikipedia (at least the release): https://en.wikipedia.org/wiki/Webots
- [ ] Update the version on the `webots` repo.
- [ ] In the `released` branch, update the urls that are not updated by `scripts/new_version/new_version.sh` (commented files).
- [ ] Check that the documentation serves the correct files. jsDelivr caches the files up to 24h so it may be necessary to purge the cache manually: https://github.com/cyberbotics/cyberbotics/blob/master/wiki/How-to-purge-jsDelivr-cache.md
- [ ] Update the other repos:
  - [ ] https://github.com/cyberbotics/webots_ros
  - [ ] https://github.com/cyberbotics/webots_ros2 (Update the tag for the CI tests on [master](https://github.com/cyberbotics/webots_ros2/blob/master/scripts/ci_before_init_embed.bash#L3) and [develop](https://github.com/cyberbotics/webots_ros2/blob/develop/scripts/ci_before_init_embed.bash#L3))
  - [ ] https://github.com/cyberbotics/urdf2webots (major releases only: the header line of the proto files should refer to the new version of Webots, create a new release with a new tag will publish the updated pip package)
  - [ ] https://github.com/cyberbotics/webots-snap (several references to the current version)
  - [ ] https://github.com/cyberbotics/webots-docker (several references to the current version)
  - [ ] https://github.com/Homebrew/homebrew-cask/blob/master/Casks/webots.rb#L2-L3
- [ ] Add a new UTM image to the `webots_ros2` wiki, in the macOS installation instructions. The new image should contain the new `webots_ros2` release, compatible with the new version of Webots (e.g. version `2023.0.0` for Webots `R2023a`).
- [ ] Update the online demos to work with the new version: https://www.cyberbotics.com/animations/soccer_example/soccer.html

gabryelreyes, 2024-07-30:
> Decide which issues must be closed for this release.

I would like to wait for #6591 as we are actively working on it. The rest that are assigned to R2024a, we can move it to R2024b

gabryelreyes, 2024-07-30:
> Update all the worlds and check for unwanted warnings by running the tests/test_worlds.py script (may take a long time).
Tests fail: #6599

gabryelreyes, 2024-07-31:
Could someone take over the QA tests for macOS? Maybe @brettle or @CoolSpy3 ? We also need help with Ubuntu 20.04 and 22.04

CoolSpy3, 2024-08-01:
Unfortunately, I don't actually have a mac to test on, so I can't help with mac QA. I could conceivably run the Ubuntu tests (in a vm), but I'll be out next week, and my schedule's going to start to get kind of crazy toward the end of the month & onwards, so if someone else is available, they could probably deliver faster results.

brettle, 2024-08-02:
> Could someone take over the QA tests for macOS? Maybe @brettle or @CoolSpy3 ? We also need help with Ubuntu 20.04 and 22.04

Sorry for the slow reply. Although I have access to a MacOS laptop, I currently have other priorities in terms of webots-related work. That said, before your comment, @CoolSpy3 and I were already talking about the possibility of assisting with a getting the R2024a release out. I think @CoolSpy3 was hoping to get #6354 and #6566 addressed first. He is currently actively working on at least #6354. All that is to say, I can help QA on MacOS, just not quite yet. If someone else can do it before I'm ready, they definitely should.

I can't QA directly on ubuntu because I'm running Fedora 40.

gabryelreyes, 2024-08-05:
Sounds good, I see there are 2 PRs open for the issues you mentioned. So I would wait for these, and then make the release at that point.

ShuffleWire, 2024-08-08:
From what I've read, it look like https://github.com/cyberbotics/webots/pull/6548 it ready to merge from some time now. Do you want to include it into the release ? Maybe ping the author if need be ?

omichel, 2024-08-08:
> From what I've read, it look like #6548 it ready to merge from some time now. Do you want to include it into the release ?

Unfortunately, it doesn't pass the tests and should be fixed if we want to include it in the release. @harunkurtdev.

harunkurtdev, 2024-08-08:
> > From what I've read, it look like #6548 it ready to merge from some time now. Do you want to include it into the release ?
> 
> Unfortunately, it doesn't pass the tests and should be fixed if we want to include it in the release. @harunkurtdev.

Sir, it is all test passed, can you  merge ?

CoolSpy3, 2025-08-17:
Closing as R2025a has been released.

closed 2025-08-17 by CoolSpy3