cyberbotics/webots #6636: Issues with Supervisor and ROS 2

miky-alt, 2024-08-20:
**Description**
When i start the simulation with `ros2 launch crazychoir_examples forest_aggregative_webots.launch.py` i get the following error in the Webots console :
```
Traceback (most recent call last):
  File "fire.py", line 23, in <module>
    from controller import Supervisor
  File "/usr/local/webots/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: /usr/local/webots/lib/controller/python38/_controller.so: undefined symbol: _ZNK6webots5Robot7getTypeEv
WARNING: 'fire' controller exited with status: 1.
```
. 
To address this, I added the Webots library path to the **LD_LIBRARY_PATH** variable with the following command:  `export LD_LIBRARY_PATH=$WEBOTS_HOME/lib/controller:$LD_LIBRARY_PATH` . 
After running echo $LD_LIBRARY_PATH, its value is:
```
echo $LD_LIBRARY_PATH
/usr/local/webots/lib/controller:/home/opt4smart/crazychoir_ws/install/webots_ros2_control/lib:/home/opt4smart/crazychoir_ws/install/webots_ros2_driver/lib:/home/opt4smart/crazychoir_ws/install/webots_ros2_msgs/lib:/home/opt4smart/crazychoir_ws/install/vicon_receiver/lib:/home/opt4smart/crazychoir_ws/install/crazychoir_interfaces/lib:/home/opt4smart/crazychoir_ws/install/choirbot_interfaces/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:/usr/local/nvidia/lib:/usr/local/nvidia/lib64:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
```
.

 Now, when I restart the simulation, I receive the following errors in the terminal: 
```
ros2 launch crazychoir_examples forest_aggregative_webots.launch.py
[INFO] [launch]: All log files can be found below /home/opt4smart/.ros/log/2024-08-19-17-15-52-518838-michelangelo-Precision-5530-22305
[INFO] [launch]: Default logging verbosity is set to INFO
No protocol specified
[INFO] [webots-1]: process started with pid [22324]
[INFO] [crazychoir_aggregative_lighthouse_gui-2]: process started with pid [22326]
[INFO] [driver-3]: process started with pid [22330]
[INFO] [robot_state_publisher-4]: process started with pid [22332]
[INFO] [crazychoir_aggregative_webots_guidance_aggregative-5]: process started with pid [22334]
[INFO] [crazychoir_aggregative_webots_simple_guidance-6]: process started with pid [22336]
[INFO] [crazychoir_aggregative_webots_controller-7]: process started with pid [22338]
[INFO] [crazychoir_aggregative_webots_trajectory-8]: process started with pid [22341]
[INFO] [driver-9]: process started with pid [22343]
[INFO] [robot_state_publisher-10]: process started with pid [22346]
[INFO] [crazychoir_aggregative_webots_guidance_aggregative-11]: process started with pid [22348]
[INFO] [crazychoir_aggregative_webots_simple_guidance-12]: process started with pid [22350]
[INFO] [crazychoir_aggregative_webots_controller-13]: process started with pid [22361]
[INFO] [crazychoir_aggregative_webots_trajectory-14]: process started with pid [22376]
[INFO] [driver-15]: process started with pid [22379]
[INFO] [robot_state_publisher-16]: process started with pid [22381]
[INFO] [crazychoir_aggregative_webots_guidance_aggregative-17]: process started with pid [22384]
[INFO] [crazychoir_aggregative_webots_simple_guidance-18]: process started with pid [22397]
[INFO] [crazychoir_aggregative_webots_controller-19]: process started with pid [22399]
[INFO] [crazychoir_aggregative_webots_trajectory-20]: process started with pid [22401]
[webots-1] Warning: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-opt4smart'
[driver-3] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 1)
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] [INFO] [1724087753.141080976] [agent_0.robot_state_publisher]: got segment 
[driver-9] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 1)
[robot_state_publisher-10] Parsing robot urdf xml string.
[robot_state_publisher-10] [INFO] [1724087753.135614468] [agent_1.robot_state_publisher]: got segment 
[driver-15] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 1)
[robot_state_publisher-16] Parsing robot urdf xml string.
[robot_state_publisher-16] [INFO] [1724087753.174439200] [agent_2.robot_state_publisher]: got segment 
[crazychoir_aggregative_webots_guidance_aggregative-5] No protocol specified
[driver-3] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 2)
[driver-9] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 2)
[driver-15] Cannot open file: /tmp/webots-22324-E8Di9E/WEBOTS_SERVER (retry count 2)
[crazychoir_aggregative_webots_simple_guidance-6] No protocol specified
[crazychoir_aggregative_webots_guidance_aggregative-11] No protocol specified
[crazychoir_aggregative_webots_simple_guidance-12] No protocol specified
[crazychoir_aggregative_webots_simple_guidance-18] No protocol specified
[crazychoir_aggregative_webots_guidance_aggregative-17] No protocol specified
[crazychoir_aggregative_webots_controller-7] [INFO] [1724087754.865656150] [agent_0.controller]: Go!
[crazychoir_aggregative_webots_controller-19] [INFO] [1724087754.894551358] [agent_2.controller]: Go!
[crazychoir_aggregative_webots_trajectory-8] [INFO] [1724087754.973515780] [agent_0.trajectory]: FullState reference 0 started
[crazychoir_aggregative_webots_trajectory-8] [WARN] [1724087754.974521011] [agent_0.trajectory]: No trajectory parameters passed by argument.
[crazychoir_aggregative_webots_trajectory-8] [WARN] [1724087754.975285701] [agent_0.trajectory]: Wait for topic: "/agent_0/traj_params"
[... 27 lines cut ...]
[driver-9]   what():  Error: The Python module with the WebotsNode class cannot be executed.
[driver-15] terminate called after throwing an instance of 'std::runtime_error'
[driver-15]   what():  Error: The Python module with the WebotsNode class cannot be executed.
[ERROR] [driver-3]: process has died [pid 22330, exit code -6, cmd '/home/opt4smart/crazychoir_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args -r __ns:=/agent_0 --params-file /tmp/launch_params_ep_2y2fz'].
[ERROR] [driver-9]: process has died [pid 22343, exit code -6, cmd '/home/opt4smart/crazychoir_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args -r __ns:=/agent_1 --params-file /tmp/launch_params_de0s7lzq'].
[ERROR] [driver-15]: process has died [pid 22379, exit code -6, cmd '/home/opt4smart/crazychoir_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args -r __ns:=/agent_2 --params-file /tmp/launch_params_0z038u8_'].

```
. As a result, the simulation does not progress.

**System**

- Operating System: Linux Ubuntu 20.04
- Graphics Card: NVIDIA Quadro P2000
- Webots R2022a
- Ros 2 foxy

**Additional context**
The simulation works in two cases. If I execute `export LD_LIBRARY_PATH=$WEBOTS_HOME/lib/controller:$LD_LIBRARY_PATH` and echo $LD_LIBRARY_PATH shows: 

```
echo $LD_LIBRARY_PATH
/usr/local/webots/lib/controller:/home/opt4smart/crazychoir_ws/install/webots_ros2_control/lib:/home/opt4smart/crazychoir_ws/install/webots_ros2_driver/lib:/home/opt4smart/crazychoir_ws/install/webots_ros2_msgs/lib:/home/opt4smart/crazychoir_ws/install/vicon_receiver/lib:/home/opt4smart/crazychoir_ws/install/crazychoir_interfaces/lib:/home/opt4smart/crazychoir_ws/install/choirbot_interfaces/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:/usr/local/nvidia/lib:/usr/local/nvidia/lib64:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
```
and I remove the ROS 2 nodes from the launch file, keeping only the part that loads the Webots world. Alternatively, it works if I reset the LD_LIBRARY_PATH variable to its previous value before `export LD_LIBRARY_PATH=$WEBOTS_HOME/lib/controller:$LD_LIBRARY_PATH` and remove the Supervisor node. But i want them to work together.

lukicdarkoo, 2024-08-28:
ROS Foxy is EOL for a while. Would it be possible for you to upgrade?

It would be helpful to share your launch file and URDF.

lukicdarkoo, 2024-11-28:
@miky-alt We don't support Foxy anymore. Please try upgrading to Humble or Jazzy. If the problem persists feel free to reopen the issue.

closed 2024-11-28 by lukicdarkoo