cyberbotics/webots #6672: Rotation and translation of the 3D viewport interface

wangkang123123, 2024-10-05:
**Describe the Bug**
In the 3D viewport, there is a problem when controlling the rotation and panning of the screen with the mouse. The panning distance is very short. When rotating, the picture rotates randomly, and the center of rotation of the picture is not on the object.

**Screenshots**

https://github.com/user-attachments/assets/b2561740-91c0-4639-90bf-1486ab9e4a01

**System**
 - Operating System: Linux Ubuntu 20.04.

lukicdarkoo, 2024-11-28:
@wangkang123123 Can you reupload the video?

ShuffleWire, 2026-04-05:
AFAIK, you need to select the object to get the rotation happening.
It should be a bug that I've ran into a lot, it look like the center of rotation should be where the cursor point to, but if nothing is there, it rotate around the camera (as when you click on the sky for instance)
But the "nearest object location' detection seems to fail sometime, and result in what you see. 

Does it help, do you have the issue still on the lastest webots ?