cyberbotics/webots #6675: Segmentation fault on Windows by following Tutorial

Fredrikhb, 2024-10-14:
**Describe the Bug**
CTD

**Steps to Reproduce**
1. Follow https://github.com/cyberbotics/webots/wiki/Windows-installation/ to install webots on Windows
2. Run webots through MINGW64 by typing 'webots' and hit enter
3. Add a Robot node having four HingeJoint nodes as children.
4. Add a Shape node to the boundingObject field of the Robot node.
5. In the appearing geometry NULL field, add a Box node.
6. Crash to desktop. 
7. Read off the MINGW64 terminal "Segmentation fault"

**Expected behavior**
A box node to be added to the geometry field of the Robot node's Shape node.

**System**
 - Operating System: Windows 10 Education 64-bit
 - Graphics Card: Intel Iris Xe Graphics

**Additional context**
I simply followed the Tutorial 6: 4-Wheels Robot.

blazaid, 2024-10-18:
Hi there,

I am experiencing the same segmentation fault described in the original post, but in my case, I'm running Webots on Linux Mint 22 "Wilma". The crash happens under the same conditions:

1. Create a "Robot" node.
2. Under "BoundingBox", add a "Shape" node.
3. Inside the "Shape", under "geometry", add a "Box" node.

Webots crashes immediately with the following error:

`/usr/local/bin/webots: line 105: 33647 Segmentation fault      (core dumped) "$webots_home/bin/webots-bin" "$@"`

**System information:**

- Linux Mint 22 "Wilma"
- Webots version: 2023b
- Graphics card: WhiskeyLake-U GT2 [UHD Graphics 620]

It seems to be the same issue as previously reported.

ShuffleWire, 2024-10-21:
Reproduced with Webots 2023b, Debian 12
Traceback
````
Thread 1 "webots-bin" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, 
    no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) where
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007f090b6a9e9f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007f090b65afb2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f090b645472 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007f090b8b9746 in dDebug(int, char const*, ...) (num=2, msg=0x7f090b9f9740 "geom must be placeable in %s()") at ode/src/error.cpp:96
#5  0x00007f090b87dd69 in dGeomSetPosition_ST(dxGeom*, double, double, double) (g=0x55922c666140, x=0, y=0, z=0) at ode/src/collision_kernel.cpp:569
#6  0x00007f090b9083c2 in dGeomSetPosition_MT(dxGeom*, double, double, double, void (*)(dxGeom*, double, double, double))
    (g=0x55922c666140, x=0, y=0, z=0, _geomSetPositionFunc=0x7f090b87dcf3 <dGeomSetPosition_ST(dxGeom*, double, double, double)>) at ode/src/ode_MT/ode_MT.cpp:501
#7  0x00007f090b87defd in dGeomSetPosition(dxGeom*, dReal, dReal, dReal) (g=0x55922c666140, x=0, y=0, z=0) at ode/src/collision_kernel.cpp:595
#8  0x000055921de333b0 in WbMatter::updateOdePlaceableGeomPosition(dxGeom*) ()
#9  0x000055921de336d8 in WbMatter::updateOdeGeomPosition(dxGeom*) ()
#10 0x00007f090e797772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#11 0x00007f090e797772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#12 0x000055921df19c4c in WbSFNode::setValue(WbNode*) ()
#13 0x000055921de79368 in WbNodeOperations::initNewNode(WbNode*, WbNode*, WbField*, int, bool, bool) ()
#14 0x000055921df0dc16 in WbSceneTree::addNew() ()
#15 0x000055921df0e0d6 in WbSceneTree::handleDoubleClickOrEnterPress() ()
#16 0x00007f090e797772 in  () at /home/yann/webots/lib/webots/libQt6Core.so.6
#17 0x00007f090d3cd383 in QWidget::event(QEvent*) () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#18 0x00007f090d461322 in QFrame::event(QEvent*) () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#19 0x00007f090e738d7a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#20 0x00007f090d376f51 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#21 0x00007f090d37f8d8 in QApplication::notify(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#22 0x00007f090e738fda in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#23 0x00007f090d37ec40 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
    at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#24 0x00007f090d3dde61 in  () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#25 0x00007f090d3e0a5b in  () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#26 0x00007f090d376f62 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Widgets.so.6
#27 0x00007f090e738fda in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#28 0x00007f090dbbd4d1 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /home/yann/webots/lib/webots/libQt6Gui.so.6
#29 0x00007f090dc0f0dc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Gui.so.6
#30 0x00007f09085a6aca in  () at /home/yann/webots/lib/webots/libQt6XcbQpa.so.6
#31 0x00007f090c31c7a9 in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007f090c31ca38 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007f090c31cacc in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007f090e9df90a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#35 0x00007f090e744c9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt6Core.so.6
#36 0x00007f090e74107e in QCoreApplication::exec() () at /home/yann/webots/lib/webots/libQt6Core.so.6
#37 0x000055921dcd4d1e in main ()
```

ShuffleWire, 2024-10-21:
For the immediate fix, considering the assert "geom must be placeable"
In the boundingObject, first add a Pose, then in the field children, add the Shape and the box, it work well, and let you move the boundingObject

It turn out that adding only the box (without the Shape node, nor the Pose) work as well.

By curiosity, adding the Shape + Box into the children fields, giving it a DEF name, then using it as the boundingObject work as well

So it's seems that something sneaky is happening here in this specific case...

CoolSpy3, 2024-10-29:
It looks like the culprit is `WbGeometry::setOdeData` (called from `WbMatter.cpp:387`). For some reason, `mOdeGeom == geom` (although I'm still not sure where this is set), so `gGeomDestroy` destroys `geom`, so `insertedGeom` (in `WbMatter`) effectively becomes a null pointer, resulting in undefined behavior. We should figure out whether the state of `mOdeGeom` is a bug. If `setOdeData` can, in fact, be called with `geom == mOdeGeom`, `WbGeometry:474` should probably be changed to read `if (mOdeGeom && mOdeGeom != geom)`. If the state is invalid, the underlying bug should be fixed, and `WbGeometry:469` should be updated to read `assert(geom && geom != mOdeGeom && matterAncestor);`

CoolSpy3, 2024-10-29:
I think I've figured it out! See the linked PR for details.

closed 2024-10-30 by CoolSpy3