cyberbotics/webots #6687: Questions Regarding GPU Acceleration for Physics Simulation in Webots

wayne-weiwei, 2024-10-26:
**Describe the Bug**
Currently, Webots relies on the CPU for physics simulations (e.g., physics engine calculations and sensor data processing), while the GPU is mainly used for graphical rendering. This results in limited performance improvement for physics calculations, even when GPU acceleration is enabled. For complex simulations, CPU-only processing is insufficient, and the lack of GPU acceleration for physics calculations is a significant bottleneck.

**Steps to Reproduce**

1. Clone the Webots repository and start Webots using ./webots.
2. Set up a complex simulation environment with multiple physics calculations (e.g., collision detection, path planning, etc.).
3. Monitor the performance and observe that GPU usage remains primarily limited to rendering tasks.
4. Observe that the CPU load remains high while physics calculations do not leverage GPU acceleration.

**Expected behavior**
I expected that Webots could leverage GPU resources for physics simulations to accelerate tasks such as collision detection and sensor data processing, allowing for faster performance in complex environments. GPU acceleration for physics simulations would significantly enhance Webots' capabilities in handling resource-intensive tasks.

ShuffleWire, 2024-11-01:
Physic in Webots relies on the ODE solver https://www.ode.org (plus some modification I believe) which is not designed at all to run on GPU. So that would be a lot of work to get it done.
Moreover the nature of the computation in Webots might not be well suited for GPU computation. A lot a I/O, varying computation type ans quantity, and each of them not being very expensive. That is, pushing it to the GPU would result in overhead greater than the possible gain in speed.
That my 2cents, I'm not a GPU dev, nor a physic engine dev. But doing so in Webots would still be a huge task, if not possible at all.

However, some sensors are GPU proceced : Lidar and Camera, if I don't miss anything.
Moreover, you should be able to multi-thread Webots by setting this the preference menu. However, I never played with that, and I believe that you need to have multiple object to take advantage of that, because one object (especially Robot and Solid) can't be multi threaded. You may have some room for improvement here. Caution that it will render your simulation non reproducible.

I do think that maybe it would be necessary to benchmark where the bottleneck is. Could it be you driver code ?
Did you read https://cyberbotics.com/doc/guide/speed-performance ? It might get you some info.

Cheers

lukicdarkoo, 2024-11-28:
I believe @ShuffleWire gave a great response. Utilzing GPU for physics would be very hard due to the nature of the physics simulation. I would try to optimize the simulation. We have fairly complex physics simulations running in real-time (e.g. https://www.youtube.com/watch?v=dysQ4EugD0s).