cyberbotics/webots #6696: Increasing memory consumption iteratively creating and deleting camera nodes

msminirobot, 2024-11-13:
**Describe the Bug**
In an evolutionary robotics setting, I create new robot nodes iteratively and delete the old ones using a supervisor node. Over time, the memory consumption increases up to very high levels (> 30 GB). Isolating the other elements of my robots, I notice the memory increase only when enabling a camera device 

`
camera = robot.getDevice('rgb_camera')
camera.enable(params.TIME_STEP)
`

**Steps to Reproduce**
1. Create robot node with camera using a supervisor node
2. Delete the robot node with the supervisor 
3. Create a new supervisor node
4. Repeat steps 1-3 iteratively, you will see the memory usage of Webots increasing over time 

**Expected behavior**
Memory consumption does not increase continuously. 

**System**
 - Operating System: macOS 15.0.1 
 - Python 3.12 

**Additional context**
Add any other context about the problem here.

austingreisman, 2024-11-13:
Sounds like a memory leak (I'm not a contributor, but use Webots). Is it possible for you to use the camera.disable() method before deconstruction to see if that "fixes" the issue?

msminirobot, 2024-11-13:
Yes, this is a memory leak. However, I don't know which part of the camera node it exactly causes. 

Using disable, deleting camera object and/or robot node doesn't help. I tried all of those combinations and my guess is that there is something not properly removed by the destructor.