cyberbotics/webots #6699: Position Sensor at medium to high speed is inaccurate and varies depending on World BasicTimeStep

austingreisman, 2024-11-15:
**Describe the Bug**
It is possible there is an issue with the Webots Position Sensor when measuring a Hinge Joint being acted upon by a RotationalMotor at high speeds on an ackermann vehicle (I used the TeslaModel 3 proto from the webots cloud platform).

While testing, converting to speed (m/s) using the Position Sensor measurements over consecutive timesteps yields values that change in accuracy overtime (never being actually accurate).

This is on Webots R2023b 

**Steps to Reproduce**
1. Open a flat Webots world with the Tesla Model.
2. Run the vehicle at a set speed.
3. Record the speed determined by taking the measurements overtime from the Position Sensor, record a GPS sensor speed output, and record your commanded speed.
4. See error.

**Expected behavior**
I would expect the RPM (converted to speed to compare) would be the same as the commanded speed. I would also expect it would match the output GPS speed.

**Screenshots**
[image]
In the above image, we have the vehicle dropping from a small height when the simulation starts (hence the weird GPS speed blip at the beginning). Then I begin to command to vehicle at various speeds and attempting to hold those speeds as the vehicle approaches it. You can see that the GPS speed, which comes directly from the gps sensor, properly matches the commanded speed. However, the reported speed, which is calculated from taking the changing value of the Position Sensor output on both front wheels over time (time being the basicTimeStep which in this case was 30 ms).
[image]
This is a steady state test, where we command a speed of 25 m/s continuously for the whole ~27 second. As seen above, the GPS reported speed approaches a steady state of 25 m/s. However, the reported speed is over predicted the whole time. 

I attempted these tests with a different wheel radius (0.28 instead of the 0.36 reported in the proto) to see if maybe that was my issue. However, the GPS data then no longer fits the commanded speed and the reported speed becomes even more incorrect.

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: Mesa Intel® Xe Graphics (TGL GT2)

**Additional context**
I attempted to find some documentation on how to properly implement measuring of the Position Sensor but was unable. I currently do this in my custom WebotsController which I supply to the Teslamodel.

Any advice or help would be greatly appreciated (I feel like this is a bug but I'd be happy to be proven wrong).

lukicdarkoo, 2024-11-28:
Can you provide a simple world and controller so we can evaluate this?

A few general (hopefully related) comments:
- Lower `basicTimeStep` means more accurate simulation overall
- GPS returns total speed in all directions, taking speed from a position sensor means speed along one axis only (you need to move the robot perfectly perpendicular to a wall)

austingreisman, 2024-12-02:
Hi @lukicdarkoo,

I've created a sample set up for your team to checkout. It can be downloaded here: https://github.com/austingreisman/simpleWebotsController

in [plotCompareSpeeds.py](https://github.com/austingreisman/simpleWebotsController/blob/main/controllers/my_controller/plotCompareSpeeds.py)

It will take the file generated by running the simulation and plot it for you to look at. Here is a sample that matches my results above.

[image]

You can look at the README for additional information.

Let me know what you find.
Thanks for the help

lukicdarkoo, 2025-01-13:
If you set the `basicTimeStep` to e.g. 1 you will see significantly lower disperity. Lower `basicTimeStep` means more accurate and stable physics.

E.g. in your example, once Tesla reaches 50/ms your `avgChangeRotationalPosition` variable reaches `2.3` radians. That is almost a half circle. Physics is just too much discretized for the speeds.

 I am not sure if we can do anything about it. It is probably the easiest to reduce the `basicTimeStep`.

austingreisman, 2025-01-14:
Hi @lukicdarkoo 

It appears at a basicTimeStep of 1, it sort of follows the GPS, but still overshoots the entire time (assuming the GPS is correct)
[image]

However, this is useable.
Here is at 10 which is also not very useable.
[image]

I'm confused how this issue relates to the physicis being too discretized, this is an issue with a sensor, it shouldn't be effected this significantly by the physics engine. I'd imagine at any speed and any timestep, the sensor would perform the same.