cyberbotics/webots #6703: Mac Senoqia  15.1 Python not support

harunkurtdev, 2024-11-17:
**Describe the Bug**

WARNING: Python was not found.
To fix the problem, you should set the full path of your python command in Webots->preferences->python command.
INFO: ardupilot_vehicle_controller: Starting controller: 
WARNING: ardupilot_vehicle_controller: failed to start: 
WARNING: Unable to find the 'python' executable in the current PATH. Please check your python installation. It should be possible to launch python from a terminal by typing 'python'. It may be necessary to add the python bin directory to your PATH environment variable. More information about the python installation is available in Webots' User guide.

I downloaded ROS2 jazzy for macOS but not working with python3 support

**Screenshots**
[image]

**System**
 - Operating System: macOS senoqia M3

harunkurtdev, 2024-11-17:
here a another example

[image]

lukicdarkoo, 2024-11-28:
@harunkurtdev What do you get when you run `/Users/harunkurtdev/.pyenv/shims/python3 --version`?

harunkurtdev, 2024-12-17:
Mr. @lukicdarkoo, it is showing  `Python 3.10.12`.

harunkurtdev, 2024-12-22:
okey sir I solve, I mistake a little.

closed 2024-12-22 by harunkurtdev