cyberbotics/webots #6728: Webots only runs when launched by the installer

Doodlinka, 2024-12-11:
**Describe the Bug**
Webots crashes at the "loading world" stage (85-100%), refusing to elaborate on Windows 10 and complaining about missing DLLs (which i found in msys4/mingw64/bin/cpp IIRC) on Windows 11 (4 error boxes). The error happened on two different machines on both user and system-wide installations (though i haven't tried system-wide on Windows 10). The crash doesn't happen when Webots is run by the installer via the final "Launch Webots" checkbox, so it can be circumvented by reinstalling the program every time you use it (but it's dumb). May be a duplicate, but all other issues seem closed yet here it is.

**Steps to Reproduce**
1. Install Webots 2022b, 2023a, or 2023b (haven't tested others).
2. On the "Setup has finished" stage, uncheck "Launch Webots", or launch it and close the program.
3. Try to start it again.

**Expected behavior**
Webots should run when started by the user.

**Screenshots**
coming later, maybe

**System**
 - Operating System: Windows 10 and 11
 - NVIDIA GeForce GTX 1660 Ti on the Windows 10 machine, coming later for Windows 11

**Additional context**
Add any other context about the problem here.

lukicdarkoo, 2024-12-11:
There are many Webots users on Windows and each release is manually tested on a clean Windows machine, so it is strange you are experiencing the issue. I am pretty sure there is something with your environment, but we have to figure out what exactly.

Could it be possible you are using an Antivirus program that's messing up with the shared libs?

Doodlinka, 2024-12-12:
My only antivirus is Windows Defender. I'll try on a nearly clean Windows machine, and a clean VM, with Defender on and off. Will also try letting it through the firewall, which someone reported as a solution to a similar problem.

Doodlinka, 2025-08-12:
i've returned to this program in version 2025a and it works for me now, so i guess it was fixed