cyberbotics/webots #6731: webots 2023b bug about segmentation_image() don't save segment image

liuzm-slam, 2024-12-16:
i use this function segmentation_image() get the segment picture and save the segment picture to my local computer path, but occur crash:     

**segmentation_image = rgb.segmentation_image()
  File "/usr/local/webots/lib/controller/python/controller/camera.py", line 168, in segmentation_image
    return bytes(wb.wb_camera_recognition_get_segmentation_image(self._tag).contents)
ValueError: NULL pointer access**

but, in webots can display the segment result:
[image]

i use code is:
    **_rgb = supervisor.getDevice("Astra rgb")
    rgb.enable(timestep)
    rgb.recognitionEnable(timestep)
    rgb.enableRecognitionSegmentation()
    segmentation_image = rgb.segmentation_image()_**

i don't know this problem, can you get me some clue?
and this code is also correct
                **_print(rgb.hasRecognitionSegmentation()) // output true
                objects = rgb.getRecognitionObjects()
                if not objects:
                    print("No objects recognized.")
                else: // output this
                    print(f"Number of recognized objects: {len(objects)}")
                    for obj in objects:
                        print(f"Object ID: {obj.id}")
                        print(f"Model name: {obj.model}")
                        print(f"Position: {obj.position}")
                        print(f"Orientation: {obj.orientation}")
                        print(f"Size: {obj.size}")
                        print(f"Position on image: {obj.position_on_image}")
                        print(f"Size on image: {obj.size_on_image}")
                        print(f"Number of colors: {obj.number_of_colors}")_**

liuzm-slam, 2024-12-16:
l also use this code https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/camera_segmentation/camera_segmentation.py 

**_if rgb.isRecognitionSegmentationEnabled() and rgb.getRecognitionSamplingPeriod() > 0:_** 
the function conditions can met, but crash

  **_File "/home/Project/Webots/controllers/uav_control.py", line 145, in main_thread
    data = rgb.getRecognitionSegmentationImage()
  File "/usr/local/webots/lib/controller/python/controller/camera.py", line 259, in getRecognitionSegmentationImage
    return self.segmentation_image
  File "/usr/local/webots/lib/controller/python/controller/camera.py", line 168, in segmentation_image
    return bytes(wb.wb_camera_recognition_get_segmentation_image(self._tag).contents)_**

CoolSpy3, 2024-12-16:
Hello!

It looks like the simulation has to take a step after enabling the camera before data becomes available. You also have to remove the parenthesis after `segmentation_image` as it is a property rather than a method (or use `getRecognitionSegmentationImage()` as in the second example). The following works for me:
``` python
rgb = supervisor.getDevice("Astra rgb")
rgb.enable(timestep)
rgb.recognitionEnable(timestep)
rgb.enableRecognitionSegmentation()
robot.step(timestep)
segmentation_image = rgb.segmentation_image
```
The additional code sample you posted also looks good to me!

liuzm-slam, 2024-12-16:
i know this problem， in the webots， i must reload the world, than run this code:
```
rgb = supervisor.getDevice("Astra rgb")
rgb.enable(timestep)
rgb.recognitionEnable(timestep)
rgb.enableRecognitionSegmentation()
robot.step(timestep)
segmentation_image = rgb.segmentation_image
```
if not reload the world, it fail. 
thanks @CoolSpy3

CoolSpy3, 2024-12-17:
Yep, you have to reload the world so that Webots runs the new code. Glad to hear you got it working! Good luck on your project :D

closed 2024-12-17 by CoolSpy3