cyberbotics/webots #6746: Load the world webots created in previous versions and crashes

LightRaing, 2024-12-25:
[worlds.zip](https://github.com/user-attachments/files/18242952/worlds.zip)
**Describe the Bug**
Loading world in the attachment through File->Open World in the webots menu bar, the program crashes during loading

**Steps to Reproduce**
1. Go to 'File'
2. Click on 'Open World '
3. Select & Load World
4. See error
5. The error message printed by the terminal is: /usr/local/bin/webots: 第 105 行： 12533 段错误               （核心已转储） "$webots_home/bin/webots-bin" "$@"

**System**
 - Operating System: [Ubuntu 22.04.5 LTS]
 - Graphics Card: [NVIDIA Corporation GA106 [GeForce RTX 3060 Lite Hash Rate] (rev a1)]
 - NVIDIA-SMI 565.57.01
 - webots: 2023b

CoolSpy3, 2024-12-27:
Confirmed! Here's a stacktrace:
``` c
0x00007ff6516d2e4c in WbRotation::x (this=0xfeeefeeefeeefefe) at maths/WbRotation.hpp:90
90	  double x() const { return mX; }
(gdb) #0  0x00007ff6516d2e4c in WbRotation::x (this=0xfeeefeeefeeefefe) at maths/WbRotation.hpp:90
#1  0x00007ff65154f7ec in WbRotation::WbRotation (this=0x1e985345970, r=...) at maths/WbRotation.hpp:37
#2  0x00007ff651387ae5 in WbSolid::applyHiddenKinematicParameters (this=0x1e98bd5e220, hkp=0x1e98bf57500, 
    backupPrevious=true) at nodes/WbSolid.cpp:388
#3  0x00007ff6513877e8 in WbSolid::restoreHiddenKinematicParameters (this=0x1e98bd5e220, map=..., 
    counter=@0xa051df870c: 5) at nodes/WbSolid.cpp:343
#4  0x00007ff65138785d in WbSolid::restoreHiddenKinematicParameters (this=0x1e98bd799e0, map=..., 
    counter=@0xa051df870c: 5) at nodes/WbSolid.cpp:349
#5  0x00007ff65138785d in WbSolid::restoreHiddenKinematicParameters (this=0x1e985896480, map=..., 
    counter=@0xa051df870c: 5) at nodes/WbSolid.cpp:349
#6  0x00007ff6513873fc in WbSolid::preFinalize (this=0x1e985896480) at nodes/WbSolid.cpp:318
#7  0x00007ff651254341 in WbGroup::preFinalize (this=0x1e9ee67f750) at nodes/WbGroup.cpp:85
#8  0x00007ff6511d451f in WbBaseNode::finalize (this=0x1e9ee67f750) at nodes/WbBaseNode.cpp:99
#9  0x00007ff651373b1e in WbSimulationWorld::WbSimulationWorld (this=0x1e9ee60e390, tokenizer=0xa051df8db0)
    at engine/WbSimulationWorld.cpp:105
#10 0x00007ff65120b9cf in WbControlledWorld::WbControlledWorld (this=0x1e9ee60e390, tokenizer=0xa051df8db0)
    at control/WbControlledWorld.cpp:44
#11 0x00007ff6511af862 in WbApplication::loadWorld (this=0x1e9dfece1e0, worldName=..., reloading=false, 
    isLoadingAfterDownload=true) at app/WbApplication.cpp:288
#12 0x00007ff6517f46af in QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1, 2>, QtPrivate::List<QString const&, bool, bool>, void, void (WbApplication::*)(QString, bool, bool)>::call(void (WbApplication::*)(QString, bool, bool), WbApplication*, void**)::{lambda()#1}::operator()() const (__closure=0xa051df93f0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#13 0x00007ff651676904 in QtPrivate::FunctorCallBase::call_internal<void, QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1, 2>, QtPrivate::List<QString const&, bool, bool>, void, void (WbApplication::*)(QString, bool, bool)>::call(void (WbApplication::*)(QString, bool, bool), WbApplication*, void**)::{lambda()#1}>(void**, QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1, 2>, QtPrivate::List<QString const&, bool, bool>, void, void (WbApplication::*)(QString, bool, bool)>::call(void (WbApplication::*)(QString, bool, bool), WbApplication*, void**)::{lambda()#1}&&) (args=0xa051df9620, fn=...)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:65
#14 0x00007ff65164c869 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1, 2>, QtPrivate::List<QString const&, bool, bool>, void, void (WbApplication::*)(QString, bool, bool)>::call (
    f=(void (WbApplication::*)(WbApplication * const, QString, bool, bool)) 0x7ff6511aecf4 <WbApplication::loadWorld(QString, bool, bool)>, o=0x1e9dfece1e0, arg=0xa051df9620) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:151
#15 0x00007ff65166cffc in QtPrivate::FunctionPointer<void (WbApplication::*)(QString, bool, bool)>::call<QtPrivate::List<QString const&, bool, bool>, void> (
    f=(void (WbApplication::*)(WbApplication * const, QString, bool, bool)) 0x7ff6511aecf4 <WbApplication::loadWorld(QString, bool, bool)>, o=0x1e9dfece1e0, arg=0xa051df9620) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:199
#16 0x00007ff651680330 in QtPrivate::QCallableObject<void (WbApplication::*)(QString, bool, bool), QtPrivate::List<QString const&, bool, bool>, void>::impl (which=1, this_=0x1e9e184fbb0, r=0x1e9dfece1e0, a=0xa051df9620, ret=0x0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:570
#17 0x00007ff96466e9ae in ?? () from /cygdrive/c/msys64/mingw64/bin/Qt6Core.dll
#18 0x00007ff6514dbe80 in WbProtoManager::worldLoadCompleted (this=0x1e9e5d76670, _t1=..., _t2=false, _t3=true)
    at build/debug/WbProtoManager.moc.cpp:187
#19 0x00007ff65132022a in WbProtoManager::loadWorld (this=0x1e9e5d76670) at vrml/WbProtoManager.cpp:418
#20 0x00007ff6517ea0db in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbProtoManager::*)()>::call(void (WbProtoManager::*)(), WbProtoManager*, void**)::{lambda()#1}::operator()() const (
    __closure=0xa051df97d0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#21 0x00007ff6516735a4 in QtPrivate::FunctorCallBase::call_internal<void, QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbProtoManager::*)()>::call(void (WbProtoManager::*)(), WbProtoManager*, void**)::{lambda()#1}>(void**, QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbProtoManager::*)()>::call(void (WbProtoManager::*)(), WbProtoManager*, void**)::{lambda()#1}&&) (args=0xa051df9978, fn=...)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:65
#22 0x00007ff651644f79 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbProtoManager::*)()>::call(void (WbProtoManager::*)(), WbProtoManager*, void**) (
    f=(void (WbProtoManager::*)(WbProtoManager * const)) 0x7ff65131fe78 <WbProtoManager::loadWorld()>, 
    o=0x1e9e5d76670, arg=0xa051df9978) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:151
[... cut ...]
#60 0x00007ff9571c2309 in ?? () from /cygdrive/c/msys64/mingw64/bin/Qt6Gui.dll
#61 0x00007ff964385ef3 in ?? () from /cygdrive/c/msys64/mingw64/bin/Qt6Core.dll
#62 0x00007ff964383bcf in ?? () from /cygdrive/c/msys64/mingw64/bin/Qt6Core.dll
#63 0x00007ff6512595b8 in WbGuiApplication::exec (this=0xa051dffa70) at gui/WbGuiApplication.cpp:292
#64 0x00007ff65142fd11 in main (argc=1, argv=0x1e9dfe5dec0) at gui/main.cpp:218
```
It looks like whatever `WbSolid.mBaseNode` is pointing to is getting deallocated at some point, but I'm not sure where.

rizqisubeno, 2024-12-31:
maybe as wbt world file using r2021a using default nue orientation on latest wbt file running for r2023b default is enu. maybe convert the world file ?

LightRaing, 2025-01-02:
Thank you very much. According to the method you mentioned, it can be successfully loaded and displayed. @rizqisubeno The cause of the program crash should be that the coordinate system was not converted. I also want to thank you for your  attention. @CoolSpy3