cyberbotics/webots #6760: Cannot install webots on arm64 architecture platforms

mojcris, 2025-01-15:
**Describe the Bug**

Cannot install webots on arm64 architectures. 

**Steps to Reproduce**
Try the installation methods described in the documentation. 

**Expected behavior**
To be able to install and use webots. 

**Screenshots**
Not applicable. 

**System**
 - Operating System: NVIDIA Jetson Orin 64GB of memory

**Additional context**
Add any other context about the problem here.

ShuffleWire, 2025-01-16:
What happens when you install it ? Do you have any error message ?

mojcris, 2025-01-16:
Yes sorry for not adding it. Here it is: 

```
make -j8
#
# * dependencies *
#
# PROTO files unchanged, no need to rebuild proto-list.xml
#
# * ode *
#
#
# * glad *
#
#
# * wren *
#
#
# * webots (core) *
#
# moc-generating build/release/WbZoom.moc.cpp
# moc-generating build/release/WbW3dStreamingServer.moc.cpp
# moc-generating build/release/WbWrenWindow.moc.cpp
# moc-generating build/release/WbWrenTextureOverlay.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbWrenRenderingContext.moc.cpp
# moc-generating build/release/WbWrenFullScreenOverlay.moc.cpp
# moc-generating build/release/WbWrenOpenGlContext.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbWrenCamera.moc.cpp
# moc-generating build/release/WbWrenAbstractResizeManipulator.moc.cpp
# moc-generating build/release/WbWrenAbstractManipulator.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbWorld.moc.cpp
# moc-generating build/release/WbWebotsUpdateManager.moc.cpp
# moc-generating build/release/WbWorldInfo.moc.cpp
# moc-generating build/release/WbWebotsUpdateDialog.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbVisualBoundingSphere.moc.cpp
# moc-generating build/release/WbView3D.moc.cpp
# moc-generating build/release/WbViewpoint.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbMultimediaStreamingServer.moc.cpp
# moc-generating build/release/WbVacuumGripper.moc.cpp
# moc-generating build/release/WbVideoRecorderDialog.moc.cpp
# moc-generating build/release/WbVideoRecorder.moc.cpp
# moc-generating build/release/WbVelocityViewer.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbVector3Editor.moc.cpp
# moc-generating build/release/WbVector2Editor.moc.cpp
# moc-generating build/release/WbVariant.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
[... 434 lines cut ...]
# moc-generating build/release/WbAnimationRecorder.moc.cpp
# moc-generating build/release/WbAltimeter.moc.cpp
# moc-generating build/release/WbAddNodeDialog.moc.cpp
# moc-generating build/release/WbAddInertiaMatrixDialog.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbAccelerometer.moc.cpp
# moc-generating build/release/WbActionManager.moc.cpp
# moc-generating build/release/WbAbstractCamera.moc.cpp
# moc-generating build/release/WbAbstractDragEvent.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
# moc-generating build/release/WbAbstractAppearance.moc.cpp
# moc-generating build/release/WbAboutBox.moc.cpp
../../bin/qt/moc: 1: Syntax error: "(" unexpected
../../bin/qt/moc: 1: Syntax error: "(" unexpected
make[1]: *** No rule to make target 'build/release/WbAboutBox.moc.cpp', needed by 'WbAboutBox.moc.o'.  Stop.
make: *** [Makefile:112: webots_target] Error 2
```

Any help is appreciated.

mojcris, 2025-01-16:
Yes I installed qt 6.5.3 successfully using the script provided in Webots repository.

ShuffleWire, 2025-01-16:
did you alter the scripts in any way ?

mojcris, 2025-01-16:
I can't find that file. I can only find these file: 
```
webots/src/webots/build/release/WbZoom.d
webots/src/webots/build/release/WbZoom.o
```

ShuffleWire, 2025-01-16:
yes, in fact the file in not generate, the error come from the moc generation...

in webots/src/webots/Makefile line 666, is the command line that fail
I do think that some of the variable of command are failing to expand correctly, thus giving a bad command, which is not executable du to syntaxe error

mojcris, 2025-01-16:
Thanks for looking into it. Any idea how we could change it to make it run correctly?

ShuffleWire, 2025-01-16:
i'm tring to display the line executed on my end, to check for error potential

ShuffleWire, 2025-01-16:
in webots/src/webots/Makefile line 666
after 	@echo "# moc-generating" $@
add 	@echo @$(MOC) $(MOC_FLAGS) $(INCLUDE:-isystem:-I) $< -o $@

does your line looks like 
@../../bin/qt/moc nodes/WbZoom.hpp -o build/release/WbZoom.moc.cpp

but I'm not sure at all I'm printing the correct thing...

ShuffleWire, 2025-01-16:
FYI, moc file for the core should be in webots/src/webots/build/release
build you need to interrupt make juste after moc generation to see them, i expect them the be deleted as they should be temporary file

ShuffleWire, 2025-01-16:
I've read on multiple forum that you should be checking what your sh interpreter is. It should be bash, but it look like dash could be used and causse trouble 
https://unix.stackexchange.com/questions/45781/shell-script-fails-syntax-error-unexpected

which OS are you using when building ?

mojcris, 2025-01-16:
I'm using Ubuntu 20.04

ShuffleWire, 2025-01-16:
have you been able to look my others message ? They could give insight on the issues :)

mojcris, 2025-01-16:
> in webots/src/webots/Makefile line 666 after `@echo "# moc-generating" $@ ` add `@echo @$(MOC) $(MOC_FLAGS) $(INCLUDE:-isystem:-I) $< -o $@`
> 
> make clean && make does your line looks like `@../../bin/qt/moc nodes/WbZoom.hpp -o build/release/WbZoom.moc.cpp`
> 
> but I'm not sure at all I'm printing the correct thing...

I get the same output: 
```
@../../bin/qt/moc nodes/WbZoom.hpp -o build/release/WbZoom.moc.cpp
```

mojcris, 2025-01-16:
> I've read on multiple forum that you should be checking what your sh interpreter is. It should be bash, but it look like dash could be used and causse trouble https://unix.stackexchange.com/questions/45781/shell-script-fails-syntax-error-unexpected
> 
> which OS are you using when building ? Alternatively, I do expect that you build using the nvidia sdk, which setup does it contains ? I not very familiar with that, but it seems something to look for.

My default sheel is bash. I checked it using `echo $SHELL`.

ShuffleWire, 2025-01-17:
are you able to run the command alone ?

ShuffleWire, 2025-01-17:
sorry to ask again, are you sure that the path to the webots folder don't contains any parenthesis ?

mojcris, 2025-01-17:
I tried running `make release -j8` from `webots/src/webots` and it gives the same errors. 

And also double checked the webots folder name and path and there's no parenthesis in there.

CoolSpy3, 2025-01-18:
I wonder if it could be the qt version you installed. Webots is not expected to be built on arm, so it doesn't install qt for arm. Could you try the following:
1. Open `$WEBOTS_HOME/scripts/install/qt_linux_installer.sh`
2. Change `QT_VERSION` to `6.8.1` (line 6)
3. Change line 8 to `aqt install-qt --outputdir ~/Qt linux_arm64 desktop ${QT_VERSION} linux_gcc_arm64 -m qtwebsockets`
4. Run `make clean`
5. Rerun the qt installer `$WEBOTS_HOME/scripts/install/qt_linux_installer.sh`
6. Try building again

mojcris, 2025-01-18:
Thanks for the reply @CoolSpy3 

After these changes now I get the following: 
```
# * webots (core) *
#
# moc-generating build/release/WbZoom.moc.cpp @echo build/release/WbZoom.moc.cpp
# moc-generating build/release/WbW3dStreamingServer.moc.cpp @echo build/release/WbW3dStreamingServer.moc.cpp
# moc-generating build/release/WbWrenWindow.moc.cpp @echo build/release/WbWrenWindow.moc.cpp
# moc-generating build/release/WbWrenTextureOverlay.moc.cpp @echo build/release/WbWrenTextureOverlay.moc.cpp
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ../../bin/qt/moc)
# moc-generating build/release/WbWrenRenderingContext.moc.cpp @echo build/release/WbWrenRenderingContext.moc.cpp
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ../../bin/qt/moc)
# moc-generating build/release/WbWrenFullScreenOverlay.moc.cpp @echo build/release/WbWrenFullScreenOverlay.moc.cpp
# moc-generating build/release/WbWrenOpenGlContext.moc.cpp @echo build/release/WbWrenOpenGlContext.moc.cpp
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ../../bin/qt/moc)
../../bin/qt/moc: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ../../bin/qt/moc)
```

And then the `make` command fails saying this: 

```
# compiling wren/WbWrenSmaa.cpp
# compiling nodes/utils/WbWrenVertexArrayFrameListener.cpp
# compiling vrml/WbWriter.cpp
make[1]: *** No rule to make target 'build/release/WbAboutBox.moc.cpp', needed by 'WbAboutBox.moc.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
# compiling core/WbPosixMemoryMappedFile.cpp
make: *** [Makefile:112: webots_target] Error 2
```

CoolSpy3, 2025-01-18:
Hmmm, that's probably something to do with the fact that Qt 6 on Ubuntu 20 is not *officially* supported, so Qt is searching for an updated version of glibc, which is only distributed with Ubuntu 22+. The first thing I would try is downgrading Qt (try the above with `QT_VERSION=6.7.3` and `QT_VERSION=6.7.0`). Unfortunately, I wasn't able to find an arm install of 6.5.3 (the default for Webots linux builds).

If that doesn't work, your best bet is probably to install Qt manually [either through the installer or an archived build or build it from source] and then figure out how to properly build Webots against it (probably some digging to find the relevant files and copy them to where Webots expects them to be).

mojcris, 2025-03-15:
Sorry for the delayed reply. Unfortunately I gave up. I might pick this up sometime later.

up200707458, 2025-03-15:
@mojcris You can give my response in #6079  a try, it's working for me (though on Fedora).

I don't know why those gcc errors occured, but even if your QT managed to "install" (it's just a download and a copy paste of parts of it into webots build directory) you'd still be missing some libraries (also placed under webots build dir) which the Makefile downloads from cyberbotics but are only provided for linux as x64.

See my answer and the fork's readme for more details.

mojcris, 2025-03-16:
Thanks @up200707458 , I will give that a shot sometime next week. Appreciate it.