cyberbotics/webots #6776: Torque Feedback doesn't work properly for Ball Joints

ralphieraccoon, 2025-02-12:
**Describe the Bug**
I've tried to add torque feedback to the `BallJoint` in the `motor3` example like so:

```c
#include <math.h>
#include <stdio.h>
#include <webots/motor.h>
#include <webots/robot.h>

int main(int argc, char **argv) {
  wb_robot_init();
  double time_step = wb_robot_get_basic_time_step();
  WbDeviceTag motor1 = wb_robot_get_device("motor 1");
  WbDeviceTag motor2 = wb_robot_get_device("motor 2");
  WbDeviceTag motor3 = wb_robot_get_device("motor 3");
  
  wb_motor_enable_torque_feedback(motor1, time_step);
  wb_motor_enable_torque_feedback(motor2, time_step);
  wb_motor_enable_torque_feedback(motor3, time_step);

  int counter = 0;
  while (wb_robot_step(time_step) != -1) {
    counter++;
    if (counter == 50)
      wb_motor_set_position(motor1, 1);
    else if (counter == 100)
      wb_motor_set_position(motor1, -1);
    else if (counter == 200)
      wb_motor_set_position(motor1, 0);
    else if (counter == 300)
      wb_motor_set_position(motor2, 1);
    else if (counter == 400)
      wb_motor_set_position(motor2, -1);
    else if (counter == 500)
      wb_motor_set_position(motor2, 0);
    else if (counter == 600)
      wb_motor_set_position(motor3, 1);
    else if (counter == 700)
      wb_motor_set_position(motor3, -1);
    else if (counter == 800)
      wb_motor_set_position(motor3, 0);
    else if (counter > 900) {
      wb_motor_set_position(motor1, cos(0.01 * counter));
      wb_motor_set_position(motor2, sin(0.012 * counter));
      wb_motor_set_position(motor3, cos(0.013 * counter));
    }
    printf("torques: [ x y z ] = [ %+.6f %+.6f %+.6f ]\n", 
    wb_motor_get_torque_feedback(motor1),
    wb_motor_get_torque_feedback(motor2),
    wb_motor_get_torque_feedback(motor3));
  }

  wb_robot_cleanup();
  return 0;
}
```

**Expected behavior**
Something similar to the `Hinge2Joint` in the `motor2` example set up in a similar fashion, with console output that has two different numbers that are proportional to the position of the joint:

```
torques: [ x y ] = [ -0.003593 -0.003932 ]
torques: [ x y ] = [ -0.003608 -0.003926 ]
torques: [ x y ] = [ -0.003623 -0.003920 ]
torques: [ x y ] = [ -0.003638 -0.003914 ]
torques: [ x y ] = [ -0.003653 -0.003908 ]
torques: [ x y ] = [ -0.003667 -0.003902 ]
```

**Actual behaviour**
Console output which has two identical numbers and one axis that always remains at zero:

```
torques: [ x y z ] = [ +0.001053 +0.001053 +0.000000 ]
torques: [ x y z ] = [ +0.000212 +0.000212 +0.000000 ]
torques: [ x y z ] = [ -0.000629 -0.000629 +0.000000 ]
torques: [ x y z ] = [ -0.001470 -0.001470 +0.000000 ]
torques: [ x y z ] = [ -0.002309 -0.002309 +0.000000 ]
torques: [ x y z ] = [ -0.003145 -0.003145 +0.000000 ]
```

**System**
 - Linux Ubuntu 24.04 (Noble Numbat)
 - Webots R2025a

ms-10-us, 2025-05-10:
I would like to work on this bug. Please assign it to me. Thanks!

ralphieraccoon, 2025-05-11:
> I would like to work on this bug. Please assign it to me. Thanks!

I don't believe that's nessecary, if you've got a fix you can make a pull request and cite this issue.

FreedGB, 2026-09-23:
Hey there. I'll like to work on that issue.

Could someone tell me where the "motor3" example project is located, or how to reproduce the bug?

omichel, 2026-09-24:
It is located here: https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/worlds/motor3.wbt.

The description of the issue should be sufficient to help you reproduce the bug.

closed 2026-09-27 by FreedGB