cyberbotics/webots #6784: ERROR: Device 'gps' was not found on robot "e-puck"

omarsaudkhalid, 2025-03-01:
**Describe the Bug**
I am having an issue with webots. When I try writing the following I get an error: gps=robot.getDevice(‘gps’) The error is webots can’t find gps. I’m trying this on epuck robot on the simulator.

**Steps to Reproduce**
1. Try compiling a controller that has the following line gps=robot.getDevice(‘gps’) obviosuly after including the relevant libraries and creating instance of Robot class

**Expected behavior**
Error: Device "gps" was not found on robot "e-puck"

**Screenshots**

[image]

**System**
 - Operating System: [e.g., Windows 10]
 - Graphics Card: [NVIDIA GeForce MX 230]
 - Processor: [AMD Ryzen 5 3500U]

rizqisubeno, 2025-03-02:
@omarsaudkhalid have you made sure you have added the device GPS on your e-puck robot ? if no then your device GPS cannot be connected sir.

omarsaudkhalid, 2025-03-04:
@rizqisubeno yes that was the mistake. I had assumed since epuck was on webots it would have GPS pre-equipped. Thanks