cyberbotics/webots #6786: Heron sink

sh-cz, 2025-03-07:
**Describe the Bug**
Hello, when using the Heron Robot in a self-configured environment, it sinks to the bottom, while a box in the same scenario does not. However, the Heron Robot works effectively in the provided sample "Ocean.wbt." How can this issue be resolved?

**Steps to Reproduce**
- Problem 
[image]

- Heron configuration
[image]

- Fluid configuration
[image]

CoolSpy3, 2025-03-08:
It looks like the robot is hardcoded to interact with a Fluid named "fluid". In your simulation, you've renamed it to "water", so the ImmersionProperties object isn't causing any interaction.

IMO, it might be a good idea to look into exposing the immersion properties in the proto file to allow users to modify this in the future.

sh-cz, 2025-03-10:
Thank you for your reply. Indeed, the issue was resolved by renaming the fluid to "fuild" to enable proper HSV functionality.