cyberbotics/webots #6790: How to drive the robot properly, looks like friction is messed up

mojcris, 2025-03-11:
Hi there, 

I'm trying to drive this tractor with a keyboard on a plain floor and it keeps sliding like in [this video](https://youtu.be/R01D1AopmLA). 

I've played with the friction parameters, but it's still sliding too much. Any ideas why this is happening? 

Thanks.

MNikoliCC, 2025-03-14:
Could you share your simulation?
Also, please specify your operating system, Webots version, and installation method.
It seems like this isn't one of the sample simulations (_boomer.wbt_)  but rather a custom one. 
Knowing more details might help in troubleshooting the sliding issue.

mojcris, 2025-03-15:
It looks like the problem was weight and friction parameter in the proto file. They were too low. It's good now.

closed 2025-03-15 by mojcris