cyberbotics/webots #6799: brake isn't working on R2025a

MoradSayed, 2025-03-27:
**Describe the Bug**
So i downloaded webots through apt and opened the **city.wbt** world and played with it a little bit, then i edited the controls to try different features and when i wanted to try the brakes it didn't work (there was no effect in the speed of the vehicle). 

So i made a test case where i moved the vehicle 70m backward for its spawn point and driven it till a specific line, then i tried it with:

1. normal stopping (caused by setting speed to 0 when I'm release the UP key)
2. braking + setting speed to 0 at the same time

and indeed there were no difference at all "brakes have no effect on the vehicle whatsoever" 
> Note: all tests where made with **autodrive** and **enable_collision_avoidance** being set to false.

**Steps to Reproduce**
1. open the city.wbt world
2. change those two functions in the original controller
```
void check_keyboard() {
  char u = 0, d = 0, r = 0, l = 0;
  int key;
  double s;

  // Read all currently pressed keys
  while ((key = wb_keyboard_get_key()) != -1) {
    switch (key) {
      case WB_KEYBOARD_UP:
        set_speed(speed + 0.5);
        u = 1;
        break;
      case WB_KEYBOARD_DOWN:
        if (wbu_driver_get_current_speed() > 1){
          wbu_driver_set_brake_intensity(1.0);
        } else {
          set_speed(speed - 0.1);
        }
        d = 1;
        break;
      case WB_KEYBOARD_RIGHT:
        set_autodrive(false);
        s = wbu_driver_get_current_speed();
        set_steering_angle(+0.5*(80.0-s)/80.0);   // 80.0 is used as a max comp speed (max_comp_s > max_s)
        r = 1;
        break;
      case WB_KEYBOARD_LEFT:
        set_autodrive(false);
        s = wbu_driver_get_current_speed();
        set_steering_angle(-0.5*(80.0-s)/80.0);   // 80.0 is used as a max comp speed (max_comp_s > max_s)
        l = 1;
        break;
      case 'A':
        set_autodrive(true);
        break;
    }
  }

  if (!autodrive) {
    if (!u && !d) {
      set_speed(0);
    }
    if (!l && !r) {
      int sign = -1*copysign(1, steering_angle);
      if (copysign(1, manual_steering + sign) == copysign(1, manual_steering)){  // if in the same side
        change_manual_steer_angle(sign * -1);  // Reset steering
      } else {  // if going to opposite side
        set_steering_angle(0);
      }
    }
  }
}
```
and 
```
void set_speed(double kmh) {
  // max speed
  if (kmh > 60.0)  // 250
    kmh = 60.0;
  else if (kmh < -20.0)
    kmh = -20.0;
    
  speed = kmh;

  printf("setting speed to %g km/h\n", kmh);
  wbu_driver_set_cruising_speed(kmh);
}
```
3. try the brakes by pressing **DOWN**

**Expected behavior**
Brakes (ie. `wbu_driver_set_brake_intensity(1)`) will not have any effect on the vehicle. 

**Screenshots**
1. Starting point with red line as the release or brake point (depending on the test case)
[image]

2. first case (normal deceleration: `set_speed(0)`) stopping point:
[image]

3. second case (brakes `wbu_driver_set_brake_intensity(1)` + noraml deceleration `set_speed(0)`) stopping point:
[image]

**System**
 - Operating System: Linux Ubuntu 22.04
 - Graphics Card: NVIDIA GeForce RTX 3060 6 GB

ShuffleWire, 2025-04-15:
AFAIK, the `Brake` device itself work, given that the sample world `brake.wbt`work fine
Could you try to simply your example down to a simpler world in the direction of making it a simple brake, until it work again ?

ShuffleWire, 2025-04-15:
From what I'm seen, the brake works,
Please try to review your code, and if you don't find anything obvious, make a simpler code, there are a lot of code not relevant for your bug.
If you struggle with that, please ask for help

MoradSayed, 2025-04-18:
Hi there, sorry for the late reply.

> make a simpler code, there are a lot of code not relevant for your bug.

Yeah, I think I overcomplicated my description of the issue. It all boils down to `wbu_driver_set_brake_intensity(1)` not working in the [city.wbt](https://cyberbotics.com/doc/automobile/city) world. which tbh I only used it as it was mentioned and used in the provided controller by Webots (but it doesn't work).

I found a workaround to this problem. By invoking the brake device of each wheel directly and setting the `.setDampingConstant` as I want. Don't know if this is okay or not, but it works! Here's a simple version of the working code:
```
from controller import Brake

class VehicleController:
    def __init__(self): 
        self.brakes = self.init_brakes()

    def init_brakes(self):
        lf = "left_front_brake"
        rf = "right_front_brake"
        lr = "left_rear_brake"
        rr = "right_rear_brake"
        brakes = [Brake(lf), Brake(rf), Brake(lr), Brake(rr)]
        return brakes
    
    def set_brake_force(self, normalized_force):
        for b in self.brakes:
            b.setDampingConstant(normalized_force*1200)

controller = VehicleController()
controller.set_brake_force(0.85)
```

Sorry for mixing between C function `wbu_driver_set_brake_intensity(1)` and my Python solution, but I just wanted to keep referring to the C version of the function as it is the one used in the city.wbt's BMW X5 controller.