cyberbotics/webots #6805: "Bad CPU type in executable" when launching "webots-controller" for robot external controller

kelvin-olaiya, 2025-04-10:
**Describe the Bug**
Launching the *webots-controller* executable, for external controllers, yields a "Bad CPU type in executable" error

**Steps to Reproduce**
1. Open terminal
2. Execute the $WEBOTS_HOME/Contents/MacOS/webots-controller binary

**Expected behavior**
The help message is displayed

**Screenshots**
[image]

**System**
 - Operating System: macOS Sequoia
 - Graphics Card: Apple GPU 16GB unified memory 

**Additional context**
The Mac has an Apple M4 chip

CoolSpy3, 2025-04-11:
It looks like [the relevant Makefile](https://github.com/cyberbotics/webots/blob/c098bdf742a295a65f5b10f99c8a1a2c483f387a/src/controller/launcher/Makefile#L46) only builds that executable for the host platform. GitHub Actions is currently configured to use an x64 runner, so arm code isn't being generated.

I tried putting together a fix at #6806, but I don't have a mac to test on. If you get the chance, could you try installing the build of that PR once it's ready and seeing if that fixes the problem. It should be available on [this page](https://github.com/cyberbotics/webots/actions/runs/14398324185?pr=6806) in about 40-60 minutes. If the build fails, I'll try to debug it tomorrow.

CoolSpy3, 2025-04-11:
Great; glad I could help! I'll mark the PR as ready-for-review. Once it's merged, the fix should start appearing in the nightly builds, and eventually, the next major release 👍

closed 2025-04-12 by CoolSpy3