cyberbotics/webots #6808: Can't get mouse state via C++ API

visionnavnick, 2025-04-15:
**Describe the Bug**
Can't get mouse state via C++ API but works well under C API.

**Usage**
`
supervisor_->getMouse()->enable(2);
supervisor_->getMouse()->enable3dPosition();

while (supervisor_->step(step_duration_) != -1) {
    const MouseState state = supervisor_->getMouse()->getState();
    std::cout << "left:" << state.left << "\n"
        << "middle:" << state.middle << "\n"
        << "right:" << state.right << "\n"
        << "u:" << state.u << "\n"
        << "v:" << state.v << "\n"
        << "x:" << state.x << "\n"
        << "y:" << state.y << "\n"
        << "z:" << state.z << "\n"
        << std::endl;
}
`

**Output**
left:0
middle:0
right:0
u:nan
v:nan
x:nan
y:nan
z:nan

**System**
 - Operating System: [Linux Ubuntu 22.04]

ShuffleWire, 2025-04-15:
PROTO : 

```
#VRML_SIM R2025a utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2025a/projects/objects/floors/protos/RectangleArena.proto"

WorldInfo {
}
Viewpoint {
  orientation 0.44097158747972237 0.35834167382873355 -0.8228823146922256 1.681659266092813
  position -0.2960531226275349 1.4256987926844917 1.7810157412302536
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RectangleArena {
}
Robot {
  controller "my_controller"
  supervisor TRUE
}
```

code cpp
```
#include <webots/Robot.hpp>
#include <webots/Mouse.hpp>
#include <webots/Supervisor.hpp>

using namespace webots;

int main(int argc, char **argv) {
  Supervisor *supervisor = new Supervisor();
  int timeStep = (int)supervisor->getBasicTimeStep();

  supervisor->getMouse()->enable(timeStep);
  supervisor->getMouse()->enable3dPosition();
  
  while (supervisor->step(timeStep) != -1) {
    const MouseState state = supervisor->getMouse()->getState();
    std::cout << "left:" << state.left << "\n"
    << "middle:" << state.middle << "\n"
    << "right:" << state.right << "\n"
    << "u:" << state.u << "\n"
    << "v:" << state.v << "\n"
    << "x:" << state.x << "\n"
    << "y:" << state.y << "\n"
    << "z:" << state.z << "\n"
    << std::endl;
  }

  delete supervisor;
  return 0;
}

```

visionnavnick, 2025-04-16:
> This is working just fine on my end.
> 
> Make sure you have a Robot in your simulation with the flag supervisor. (seems to work even without) Does it help ? if so maybe some error message and doc should be added to fix/document this behavior
> 
> Be careful with your enable() parameter, is it really your time step ? (even if it should be an issue here)
> 
> CAUTION: The first time you call getState the value will always be nan, wait to get the next values

Thanks for your replay.

I tested the cpp you provided and it worked quiet well.

So I check my code and find the reason.

```#include <webots/Robot.hpp>
#include <webots/Supervisor.hpp>
#include <webots/Mouse.hpp>

using namespace webots;

int main(int argc, char** argv) {
    Supervisor* supervisor = new Supervisor();
    int timeStep = (int)supervisor->getBasicTimeStep();

    supervisor->getMouse()->enable(timeStep);
    supervisor->getMouse()->enable3dPosition();

    while (supervisor->step(timeStep) != -1) {
         // I called some Field functions before reading the mouse 
        auto t = supervisor->getSelf()->getField("name")->getSFString(); state 
        std::cout << "t:" << t << std::endl;

        const MouseState state = supervisor->getMouse()->getState();
        std::cout << "left:" << state.left << "\n"
            << "middle:" << state.middle << "\n"
            << "right:" << state.right << "\n"
            << "u:" << state.u << "\n"
            << "v:" << state.v << "\n"
            << "x:" << state.x << "\n"
            << "y:" << state.y << "\n"
            << "z:" << state.z << "\n"
            << std::endl;
    }

    delete supervisor;
    return 0;
}
```
**output:**
```
t:P15_1
left:0
middle:0
right:0
u:nan
v:nan
x:nan
y:nan
z:nan

t:P15_1
left:0
middle:0
right:0
u:0.876015
v:0.495283
x:89.0392
y:48.9943
z:0.00461203

visionnavnick, 2025-04-16:
> Nice to hear ! however, I don't think you need to read any field beforehand to make it work. It should work just fine, but indeed, as you show in your log, the first set of value will **always** be nan Feel free to close the issue, if you are comfortable with this solution

Sorry for the unclear reply earlier! 

Here’s the issue: When I call the Field function before reading the mouse state, the state sometimes fails to read. But if I remove that Field call, it works perfectly every time.

ShuffleWire, 2025-04-16:
Yes reproduced here as well

ShuffleWire, 2025-04-16:
My C code isn't working anymore, that is bad...