cyberbotics/webots #6879: typo in node addTorque

sean85914, 2025-08-19:
**Describe the Bug**
As #6345, `addTorque` and `addForce` results in the same behavior using Python API.  
After digging into the code, I notice [that](https://github.com/cyberbotics/webots/blob/master/lib/controller/python/controller/node.py#L225)  
```python
        wb.wb_supervisor_node_add_force(self._ref, (ctypes.c_double * 4)(*torque), 1 if relative else 0)
```

should change to
```python
        wb.wb_supervisor_node_add_torque(self._ref, (ctypes.c_double * 3)(*torque), 1 if relative else 0)
```

CoolSpy3, 2025-08-19:
Ahh; I was wondering what was causing that issue the other day! Totally didn't think to check if it was a bug in the Python API specifically :P (Side note: It looks like `addForceWithOffset` is broken in a similar way.) Gonna have to do a bit of checking because @ShuffleWire's example is in C (although I'm guessing it might just be that the force is acting as a torque due to the hinge). Great find; I'll put in a PR later :D

CoolSpy3, 2025-08-20:
After some investigation, this is (in fact) a duplicate of #6345. (I've posted a more in-depth explanation there.)

closed 2025-08-20 by CoolSpy3